Ultimaker/Cura #14385: Slicing failed with unexpected result...

bachmarc, 2023-01-26:
### Application Version

5.3 xmas 

### Platform

Windows 10

### Printer

SOVOL SV6 

### Reproduction steps

I tried to slice a STL exported from FreeCad

### Actual results

It sliced a looooong time and than aborded 

### Expected results

A slice with tree supports

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[SS_deckel.zip](https://github.com/Ultimaker/Cura/files/10507687/SS_deckel.zip)
[deckel.zip](https://github.com/Ultimaker/Cura/files/10507693/deckel.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/10507698/cura.log)

Janitortimm420, 2023-01-27:
having this same issue right now as well

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker