Ultimaker/Cura #14387: Wipe nozzle between layers does not wipe on a tool change

sheetcam, 2023-01-26:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Custom

### Reproduction steps

On a machine with multiple extruders and 'Wipe nozzle between layers' is enabled, the nozzle is not wiped if the tool is changed. On multi-extruder machines, aguably wipe between tool changes is more important than wipe between layers.

### Actual results

The current sequence is:
Select tool 1
print tool 1 layer 1
Select tool 2
print tool 2 layer 1
wipe
print tool 2 layer 2
Select tool 1
print tool 1 layer 2
wipe
print tool 1 layer 3
...and so on

### Expected results

The preferred sequence would be:
Select tool 1
wipe
print tool 1 layer 1
Select tool 2
wipe
print tool 2 layer 1
wipe (to be honest, i'd prefer an option not to have this one)
print tool 2 layer 2
Select tool 1
wipe
print tool 1 layer 2
wipe
print tool 1 layer 3
...and so on

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Attached is a zipped 3mf that demonstrates the issue.
[KKN3I_test.zip](https://github.com/Ultimaker/Cura/files/10509526/KKN3I_test.zip)
By the way, it would be handy if you allowed 3mf attachments.

MariMakes, 2023-02-03:
Hey @sheetcam,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

This is outside my field of expertise since UltiMaker doesn't have any printers with tool changes.
I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. Fingers crossed 🤞 

You can also ask your question on our Community forum, [like this user](https://community.ultimaker.com/topic/42479-no-retraction-on-multi-extruder-nozzle-switch-is-causing-blobs/) there might be more printer modders there who can help you out.

sheetcam, 2023-02-03:
Thanks for the reply. Would this not also apply to Ultimaker's dual extrusion? When switching extruders the extruder you are switching to will have oozed a bit and would benefit from a wipe before being used. If you use a purge tower it should catch most of the gunk but I found with PETG the tower gets very hairy and had a tendency to get knocked over.

It's not actually a major problem for me. I just changed my Klipper config to wipe on all tool changes and disabled wipe in Cura. My current wipe sequence purges a short amount of filament into the purge bucket, wipes and goes straight to work without using a purge tower. It probably wouldn't work for multi colour but I mainly use dual material e.g PLA support for PETG.

rijkvanmanen, 2023-02-06:
Hey @sheetcam,

Thanks for you comment. For Ultimaker printers the wipe feature is not used, since we don't have wipers in our printers. The idea of the feature is too wipe between layers as described here: https://github.com/Ultimaker/Cura/pull/5356.

I agree with you that wiping on all tool changes makes more sense then wiping between layers. And I also think that wiping should be handled by firmware, not by Cura. So I think you chose the right approach.

Of course feel free to contribute to the feature and extent it to also wipe on a tool change 💪

closed 2023-02-06 by rijkvanmanen