Ultimaker/Cura #14392: Slicing failed multiple times

Janitortimm420, 2023-01-27:
### Application Version

5.3.0

### Platform

windows 11

### Printer

Creality cr10 v3

### Reproduction steps

1. i tried to slice with tree supports (i need them) failed with bug
2. I then tried without tree supports and it sliced 
3. I then tried cura 5.2.1 and it worked with both supports and without
4. went back to cura 5.3.0 and tried 2 more times and it failed

### Actual results

keeps failing 

### Expected results

It should slice with tree supports

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CCR10_Alex-Terrible-rework-mask.zip](https://github.com/Ultimaker/Cura/files/10515717/CCR10_Alex-Terrible-rework-mask.zip)

N6REJ, 2023-01-27:
I had a similar issue with 5.3.0-alpha-xmas.  When trying to slice a very complex project it would fail.  If I simplified it then it would succeed.
in my case it was a large object with a couple of dozen support blockers and 2 large infill-mesh objects.
This is a big file, approximately 121mb.
My system is pretty decent so shouldn't have had issues...
. cpu: amd Ryzen 5 5600x
. MB:  MSI X570S carbon max wifi
. ram: 32gb dd4 3600
. disk: 1TB pcie 4 nvme ssd

DerGenaue, 2023-02-02:
How did the fail look like?
Did it crash (error message) or did it get stuck forever?

Depending on this, this could be one of the following bugs:
#14133 (you should disable `limit branch reach` to fix this one)
#14113
Please update this issue whether one of those solutions worked for you

N6REJ, 2023-02-03:
just said there was an error slicing.

MariMakes, 2023-02-03:
Hey @Janitortimm420,

Thanks for your report 👍
This is a pretty cool model!

It seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13643
That one has been resolved. 
I can slice it in the upcoming Cura release. So keep an eye out for that. 
[image]

I'll close the issue. 

@N6REJ if you are still stuck, I suggest making a new ticket. so we can treat is separately 👍

closed 2023-02-03 by MariMakes

DerGenaue, 2023-02-03:
Pretty sure it is not https://github.com/Ultimaker/Cura/issues/13643 but rather https://github.com/Ultimaker/Cura/issues/14113
https://github.com/Ultimaker/Cura/issues/14113 should also be resolved in the upcoming release by the fixes that Jellespiker did to Arachne; so all good, though.

Janitortimm420, 2023-02-03:
Awesome. I’ll Be looking for it soon! Thank you guys.