Ultimaker/Cura #14395: Crash when opening materials preferences

sheetcam, 2023-01-27:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Custom

### Reproduction steps

Select Preferences->configure Cura->Materials

### Actual results

Cura crashes. Traceback and logs attached below.

### Expected results

Materials settings open.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[logs.txt](https://github.com/Ultimaker/Cura/files/10519833/logs.txt)
[traceback.txt](https://github.com/Ultimaker/Cura/files/10519834/traceback.txt)

MariMakes, 2023-02-03:
Hey @sheetcam,

Thanks for your report 👍
Sorry to hear your Cura crashed.

Could it be that you changed settings in your config folder?
Do you mind sharing it so I can take a look?

You can share your configuration folder by:
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version
You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

sheetcam, 2023-02-03:
After further investigation, I have a couple of printers defined and the crash only happens with one of them. I have attached my config directory.
The offending printer profile is 'Hictop', which is a modified Koonovo IDEX configuration. Sidewinder X1 seems to work as expected.

[5.2.zip](https://github.com/Ultimaker/Cura/files/10580097/5.2.zip)

MariMakes, 2023-02-13:
Hey @sheetcam,

Thanks for sharing the files.
I can reproduce the crash, and it definetly seems to be connected to the Hictop/Koonovo printer.
[image]

I took a look but I cannot figure out what is happening. 😖 
I'll have one of the developers take a look. 👍

MafinBoy, 2024-02-17:
I just got the exact same error.
Freshly installed cura with the Sovol SV01 printer preset.
Only changed start gcode, nothing else.
[image]
[traceback.txt](https://github.com/Ultimaker/Cura/files/14319373/traceback.txt)
[logs.txt](https://github.com/Ultimaker/Cura/files/14319374/logs.txt)