Ultimaker/Cura #14397: Crashing on Preview

Talentless67, 2023-01-27:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Ender 5 Pro

### Reproduction steps

load a largeish file for comparison, once sliced it will be a 7-day print, slices fine, but when preview is selected, it gets to 99% then a white screen then the app crashes.

### Actual results

the app crashes

### Expected results

i should see a preview
[cura.log](https://github.com/Ultimaker/Cura/files/10521643/cura.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

attached is the log file

MariMakes, 2023-02-03:
Hey @Talentless67,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 
I'm having some trouble finding your preview crash in your logs. 

Do you have a project file for us that caused your preview to crash? 
It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

Talentless67, 2023-02-07:
Thanks

I didn’t load the file as it was a file I paid for, and I was not sure who would have access to it, and I didn’t want to be in breach of the agreement not to share the file

Regards

MariMakes, 2023-02-13:
Hey @Talentless67 ,

Thanks for getting back to us. 
I understand if you don't want to share a file if you paid for it. 
But I don't know how to troubleshoot your issue without the file.  🤔 

Do you have another file that reproduces the crash in the preview? 
Did you report your issue to the creator of the file? 
Perhaps they know what is going wrong. 🤷 

I'll be closing the ticket, I hope you understand.

closed 2023-02-13 by MariMakes