Ultimaker/Cura #14398: Slicing Error

dankelly2, 2023-01-27:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Creality Ender 3 Max

### Reproduction steps

I bought a file and downloaded and was printing a model bridge that will print all the parts except for 1.  When I slice it, i get an error message.  The Z height is 104.02mm, if I reduce the 100% setting to 99%, it will slice it.  My machine will print 300x300x340, so I don't know why the Z height is tripping an error.  
I called Creality and we went through some settings but were unable to figure out why it won't slice the 104.02 height.  The other piece I printed was 102.06 and it printed fine.  

Even tried resetting the printer height in the print profile and it made no difference. 

Any help would be appreciated.  my printer is a creality3 max neo

### Actual results

I'm getting an error code when I try to slice.

### Expected results

To be able to slice the file and print.
[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]

GregValiant, 2023-01-27:
Thanks for the report.
There is a bug that will keep a good model from slicing when it's at the midpoint of the Cura build plate.  Try moving and/or rotating the model and it might fool the bug.
Today a fix for the bug was rolled into Cura so (hopefully) this will be fixed in the next release.

Spacenoidx, 2023-01-27:
> Thanks for the report. There is a bug that will keep a good model from slicing when it's at the midpoint of the Cura build plate. Try moving and/or rotating the model and it might fool the bug. Today a fix for the bug was rolled into Cura so (hopefully) this will be fixed in the next release.

This bug appears even when the model is rotated and away from the midpoint of the plate.

dankelly2, 2023-01-28:
Thank You………..     I moved the model 2mm to the rear of the print bed and it seemed to slice fine.
The model company said to  make sure the support was turned off. I turn the generate  support  setting on and it also
Sliced the model fine.
Thanks for ending a 2 full day fiasco…
Dan