Ultimaker/Cura #14399: Multiple display issue  (2 lg ultragear 24", 1 vizio tv 45")

poorbullfrog, 2023-01-27:
### Application Version

5.2.1

### Platform

Windows 10 pro

### Printer

Ender 3 pro with klipper

### Reproduction steps

I had made changes to cura preferences to remove the "restore window position on restart."
Open Cura 
Open the marketplace or send any print to the printer using moonraker

### Actual results

Cura hangs, and a force quit is required. The issue is resolved if I turn off one LG monitor leaving one LG monitor and the Vizio TV on,

### Expected results

Cura not to hang and 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10523404/cura.log)

MariMakes, 2023-02-03:
HEy @poorbullfrog,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

We've received a number of reports regarding multiple monitors, and collected some workarounds. 
Do any of these help you out? 
	1. Do you have GPU Tweak OSD overlay enabled? Does Cura start if you disable that software?
	2. Do you have a Ryzen Graphics card? Can you try this workaround? https://support.makerbot.com/s/article/1667412437627
	3. Do you have the Nahimic audio enhancement service enabled? Can you try this workaround? https://github.com/Ultimaker/Cura/issues/12102#issuecomment-1157907665
       4. Do you have your monitors placed above each other? Can you try this workaround? 
       https://community.ultimaker.com/topic/42633-cura-522-stopped-working/#comment-318105

roeulogy, 2023-02-05:
I can confirm after fighting this for a few days, Asus GPU Tweak III's OSD caused this issue for me, killed the process and cura is back in action.

MariMakes, 2023-02-13:
Hey @roeulogy,

Thanks for getting back to us 👍 
Glad your issue has been resolved. 
I'll close the ticket 

Happy printing!

MariMakes, 2023-02-13:
Woops, I'm opening it again. 

@poorbullfrog you originally submitted the ticket.
Has your issue been resolved?

poorbullfrog, 2023-02-13:
Unfortunately, I have not been able to locate the program Asus GPU Tweak
III to be able to stop any processes.

MariMakes, 2023-03-31:
Quick update on our side,

We might have a fix for some of the issues reported here. 🎉
It's a bit hard for us to test, but we have a nightly build available.
Can you please try it out and let us know if it resolves your issue?

[You can find the download link and discussion here. ](https://github.com/Ultimaker/Cura/discussions/15098)

poorbullfrog, 2023-03-31:
Hello.
 I greatly appreciate the continued work being done on Cura. Unfortunately,
I can no longer reproduce the problem as I upgraded to a larger display to
replace two monitors.