Ultimaker/Cura #14400: Cant Remove Printer Profile Settings Manage Printers

cpozzij, 2023-01-28:
### Application Version

5.2.2

### Platform

Mac Os 13

### Printer

Creality Cr 10

### Reproduction steps

I Just cant remove the printer in settings, manage printers

### Actual results

i cant

### Expected results

i want to remove my old printer

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2023-01-28:
Thanks for the report.
You can't delete the "active printer" so I think you will have to install another printer, "activate" the new one, and then you can delete the old one.  The same is true for custom settings profiles.
It's expected behavior and keeps people from inadvertently deleting their printer (or profile).  I've removed the bug label but if you can't delete it with a different printer active let us know and I can stick the bug label back on.

cpozzij, 2023-01-28:
I feel so Dumb, thank you so much Greg!.
Do You know why I can’t add my printer via wi fi? Y already tried resetting everything and is on the same network and everything. Is a Creality Sermon V1. I don’t know if you can help me with that?

GregValiant, 2023-01-28:
What I don't know about WiFi fills libraries.
This site is really an Ultimaker site so I don't know if anyone here can help since it's a Creality machine.  @Asterchades may have something to add.  You might do better (couldn't do much worse) over on Reddit where there is a [Sermoon Group](https://www.reddit.com/r/Creality_Sermoon_V1/).
Good luck with it.

Asterchades, 2023-01-28:
My understanding is that Cura's built-in WiFi support only extends as far as Ultimaker printers. Unlike USB communication, which uses the rather old (and simple) UART standard, there's no real standard for talking wirelessly to 3D printers so it would require a dedicated "branch" for every brand. That's likely getting beyond the scope of the base Cura project.

Likewise I suspect that the WiFi functionality on the Sermoon series was only really intended to operate with Creality's cloud software.

There are plugins to add WiFi functionality for other printers to Cura, including OctoPrint (and Mainsail by extension) - just none specifically for the Creality printers. The MKS plugin _may_ work depending on how Creality implemented their connection, given I suspect both are using some kind of TCP/IP to UART translation layer via what is likely the same WiFi chip, but if that doesn't work you'll have to go back to an SD card or Creality's cloud software.