Ultimaker/Cura #14401: Cura allows setting infill density about 100%

branonm, 2023-01-28:
### Application Version

5.2.2

### Platform

OSX

### Printer

Ender 3

### Reproduction steps

I was intending to print something at 100% infill. I started the print as normal and noticed it was printing a big gloopy mess. I checked my settings and noticed I typed 1000 in the Infill Density dialog

### Actual results

Apparently the slicer tries to print 10x more plastic that would normally be used to print a solid infill layer.

### Expected results

I would have expected at least a warning when slicing.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I don't think any additional information is necessary. This one is pretty easy to reproduce, but I can supply photos of the results if required

GregValiant, 2023-01-28:
Thanks for the report.
I can confirm that it will take a huge number.  The fdmprinter.def.json file has a "maximum_value_warning" but no "maximum value".

@MariMakes this one should be a quick fix.

branonm, 2023-01-30:
> Thanks for flagging this 👍
> I'll bring it up with the team.
>
> I happen to know that certain materials, like the BASF Ultrafuse 17-4PH,
> have an infill sparse density that is higher than 100 because of the
> consistency of the material it causes a lot of friction to extrude the
> material an the manufacturer chose this as a way to compensate for that.

I was wondering if there were certain circumstances where an infill density
of more than 100% made sense. Thanks for the share!

MariMakes, 2023-03-13:
Hey @branonm,

We discussed your remark again with the team and came to the conclusion that we don't want to restrict people from changing the settings unless it's not slicing or breaking their printer. There are always scenarios that we forget to take into account like this one: https://github.com/Ultimaker/Cura/issues/14812

Some of the settings that @GregValiant mentioned already have another fail-safe or at least a warning indicating that the setting is not realistic. 
[image]

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Thanks again for the report! 
I'll close this issue, I hope you understand.

closed 2023-03-13 by MariMakes