Ultimaker/Cura #14403: UltimakerAccount Not signed in Please allow the application to access the requested data. You can do this by signing in again and clicking Allow

dime5800, 2023-01-28:
### Application Version

5.2.2

### Platform

10

### Printer

ender

### Reproduction steps

UltimakerAccount Not signed in Please allow the application to access the requested data. You can do this by signing in again and clicking Allow.
nothing helps

### Actual results

UltimakerAccount Not signed in Please allow the application to access the requested data. You can do this by signing in again and clicking Allow

### Expected results

UltimakerAccount Not signed in Please allow the application to access the requested data. You can do this by signing in again and clicking Allow

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

UltimakerAccount Not signed in Please allow the application to access the requested data. You can do this by signing in again and clicking Allow

MariMakes, 2023-02-03:
Hey @dime5800,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for reaching out.
Your request is unclear.
How can we help?

Do you have screenshot for us or correct steps to reproduce?

dime5800, 2023-02-04:
hello, I can't log in to the application. I click log in, the site opens, but after entering the password and login, it says that I am not logged in

-- Отправлено из мобильного приложения Яндекс Почты

MariMakes, 2023-02-10:
Hey @dime5800,

Could it be related to this ticket? 
https://github.com/Ultimaker/Cura/issues/13823

If that's not it, can you share your configuration folder? 

You can share your configuration folder by:
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version
You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-03-03 by github-actions[bot]

AAromatnyyGashish, 2025-02-12:
I'm afraid I have the same problem. if I provide the necessary data, can you try to solve my problem? could this be due to the fact that my location is in Russia?