Ultimaker/Cura #14404: Prime Tower issuue

Palacinka112, 2023-01-28:
### Application Version

5.2.2

### Platform

Windows 10

### Printer

Lotmaxx Shark V2

### Reproduction steps

When I want to print my first bicolor print with the help of the prime tower, the color change occurs at layer 29. The issue is that the PT prints layer 29 with the old color and only prints with the second color in the middle. Is there any way to make Cura print PT on the top of the old color instead of printing only to the middle of the PT with the new color? 

### Actual results

It prints with the old color PT even in layer 29 (not needed) and then prints PT with the new color in the middle, in the air, where it doesn't print anything.

### Expected results

The old color should be printed on layer 28 on the PT; then, it should change to the new color and print the whole of layer 29 with the new color. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Both materials are PLA. The first color is orange; the second one is blue.

GregValiant, 2023-01-28:
Thanks for the report.
This has come up before.  See #5282 and #10885.  It looks like this was deferred.
One of the arguments seemed to be in regards to what would happen if you have a non-compatible material in one of the extruders.  If it was a support material then it might not stick to material 1 and it would make a mess and get dragged around the bed.

Since you are only at layer 29 then a workaround could be to bring in a support blocker, make it 29 layers tall, and set it to "the new color" extruder, In the "Per Model" settings set it to print as a "Normal Model" and with walls only.  That would force the "new color" to prime on every layer.  It would probably not be a good workaround if the second extruder didn't start until layer 600.  That would waste a lot of material and time.
I'm going to remove the bug label as this is expected behavior.  If you think that is premature let me know and I'll put it back on.  It does appear that a lot of thought has gone into making it the way it is.

Palacinka112, 2023-01-29:
The non-compatible material argument doesn't help with this. I tried setting both materials as same and it did the same thing (PT for the second extruder was printed in the middle - in the air). I've attached pictures below. How do I make support for the PT? I tried it yesterday (generate support) and nothing showed in the preview and the time didn't change too. Since you are saying it is expected behavior, is there any chance how to make Cura **print the PT layer 29 only with the new color (without printing PT with the old one)**? The only solution I have found is to change extruders for the parts, but then it prints the whole PT for the second color inside the PT of the first one (i.e. PT consists of 10 layers, where 5 outside are with the old color and 5 inside are with the new one), but this seems too much of changing filament by the extruder and maybe some new issues (higher size and volume value for the PT). The best solution for me would be the one written in bold text. 
[image]
[image]
[image]

Palacinka112, 2023-01-29:
Ok, to tell the truth, I resigned in this way, so I switched the extruders (this way the PT looks like it could work, but with new Cura issues...). The new issue is that the PT prints the whole first layer with the color 2, but the object should be printed with the color 1. How to make Cura switch to color 1 on the PT layer 1?

GregValiant, 2023-01-29:
Check your Starting Gcode and see if there is an explicit call for either tool in there.  Cura always adds the "first tool" before the StartUp gcode but if there is a specific "T0" in the StartUp (possibly for heating or purge) then the switch is made and a print will always start with the tool that the StartUp gcode ends with (T0 or T1).
You could try adding **T{initial_extruder_nr}** as the last line in your StartUp gcode and see if that would fix it.  You wouldn't have to print it to check.  Just create the gcode and open the file in a text editor and take a look.  Be careful that you don't interfere with the heating order of the extruders in your StartUp or you could start with a cold extruder.

Palacinka112, 2023-01-29:
This is the beginning of the gcode. If you want to see, what can be wrong with it, I can send you the whole gcode. 
[image]

GregValiant, 2023-01-29:
That one will start with T0 (inserted by Cura right under "generated with..." and the M104 and M109 lines are for the "active extruder" which in this case would be T0.  There is no call to "T1" and so it doesn't heat.
BUT if that same print started with T1, and you didn't change the "bed plate adhesion extruder" to T1 then T0 would print part of the skirt/brim and spend the rest of the print at Standby Temp.

There are several conditions that have to be accounted for somehow:
1. Extruder 1 is the only extruder used and extruder 2 is enabled.
2. Extruder 2 is the only extruder used and extruder 1 is enabled.
3. Extruder 1 is the only extruder used and extruder 2 is DISabled.
4. Extruder 2 is the only extruder used and extruder 1 is DISabled.
5. Both extruders are used and the print starts with extruder 1.
6. Both extruders are used and the print starts with extruder 2.

Additionally, the conditions are slightly different if the printer has a "multiple-in-one-out" nozzle rather than two separate hot ends/nozzles mounted side-by-side.

A general rule would be:  If the print is only using a single extruder then disable the other extruder.  That way it won't be called to print part of the skirt/brim.  In the case of separate extruders/nozzles - the unused extruder won't spend the print at Standby Temperature.
It can be tedious to get it right.  Once you have a system in place then stick with it and you'll be fine.

Drawork, 2023-02-15:
I've been thinking of something similar, I've been working for a while on it and I think that would be awesome if there is a "check" in the dual extrusion options to make only one purge by layer, due to it's not necessary to purge twice by layer, Examples:

If you come from A and now everything is B
Prints and finish A> PT B material > prints B material > if there is more color change, print more purge layers, if not, no more 

If you come from A and then every layer has A and B materials:
Prints and finish A material> (new layer) prints the A meterial>purgue B material> prints the B material> (new layer) prints B material > PT A material > prints and finish layer A material (AND REPEAT) >> There is only one purge by layer, because on purge is the end of the last layer and the beginning fo the next one, and the purge is in the middle of the process of each layer.

Currently, it purges 3 times per two layers in parts that are 2 colors, while it can be optimized to purge once by layer if the materials are appropriate for it, and letting the user choose what to do or not would be awesome I think

Best regards