Ultimaker/Cura #14406: slicing failed

marek96681, 2023-01-28:
### Application Version

5.2.2

### Platform

Windows 11 pro

### Printer

Ultimaker S5

### Reproduction steps

1. clear build plate in Cura
2. add new parts for print
3. Change new spool in printer (Ultimaker ABS Grey)

### Actual results

Error message
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/10527244/cura.log)

### Expected results

Get slicing Preview.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I do not use NFC reader.

GregValiant, 2023-01-28:
Thanks for the report.
I didn't have any trouble slicing it.  Someone from the UM Cura team will take a look.
[image]

MariMakes, 2023-02-03:
Hey @marek96681,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

The file also slices for me. 
[image]

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
Perhaps you changed more settings than just the wall line count. 

To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-02-18 by github-actions[bot]