Ultimaker/Cura #14407: Cant slice my 3D object

Czspecter, 2023-01-28:
### Application Version

5.2.2

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

Trying to slice a 50x50x50 cube with a cuttlebone like structure. Ticked "bridge settings" but happend even without doing anything before.

### Actual results

Error message

### Expected results

Slice my 3D object correctly.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Cuttlebone_error.zip](https://github.com/Ultimaker/Cura/files/10527464/Cuttlebone_error.zip)

GregValiant, 2023-01-28:
Thanks for the report.
If you load the Mesh Tools plugin from the MarketPlace you would receive the warning that "The model is not watertight and may not print correctly".  That means the model has issues.
This is from [THIS](https://formware.co/OnlineStlRepair) site
-> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
**--> 34 Non-manifold edges (?)
--> 54 Inverted faces (?)**
--> 0 Degenerate faces (?)
**--> 8 Duplicate faces (?)**
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 97663ms ------
-> Vertex count changed from 165464 to 166730 **(+1266)**
-> Triangle count changed from 330804 to 333424 **(+2620)**

The model in the file you posted wouldn't slice.  I exported it and repaired it and it sliced with no problem.
[image]

I'm going to remove the bug label from this and close it as this problem is the model.  I hope you understand.  If you feel that is premature let us know and it can be re-opened.

closed 2023-01-28 by GregValiant

Czspecter, 2023-01-28:
> 

Thank you for your help! 
I was having trouble in Solidworks, so there must be a problem there as well.

GregValiant, 2023-01-28:
It's a complicated little model.  Make sure that there are no gaps between any features when you do the unions.