Ultimaker/Cura #14412: Pause at Height, Disarm timeout set at "0" doesn't override "#define DEFAULT_STEPPER_DEACTIVE_TIME 120"

engengr, 2023-01-29:
### Application Version

5.2.1

### Platform

WIndows 11 Pro

### Printer

Custom (Aquila X2)

### Reproduction steps

In the Post Processing Scripts for "Pause at Height", there is a "Disarm timeout" feature. If you set a time in this field for example 300, it adds a line of g-code "M18 S300" which keeps the steppers from disabling for 300 seconds. As long as I get to the printer and press resume before the steppers time out, then the resume feature functions as expected. However, if I am late and the steppers timeout and disable, then when I resume, the gantry moves up and the extruder prints too high above the model. If I select "0" in the Disarm timeout field, then the steppers timeout at the default setting specified in the Configuration.adv.h file which for my configuration is "#define DEFAULT_STEPPER_DEACTIVE_TIME 120".  If I manually add the g-code line M18 S0 to my slicer file, this overrides the default 120 seconds and keeps the steppers enabled, thus allowing the resume function to work. In my opinion, if you set 0 in the "Disarm timeout" field in Cura, the Post Processing Script should add "M18 S0" to the slicer file. 

[image]

### Actual results

If I put a value in Disarm timeout other than 0, then it adds the appropriate G-code to the slicer file.  If I leave 0 in the field, it adds nothing and the steppers disable per the configuration specified in configuration.adv.h.

### Expected results

If 0 appears in the Disarm timeout field, I think the script should add "M18 S0" to the slicer file to override the default time specified in the firmware configuration file so the steppers don't disable and the resume function works as expected.

[image]
[image]
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[10mmX1mm Disc.zip](https://github.com/Ultimaker/Cura/files/10528814/10mmX1mm.Disc.zip)

GregValiant, 2023-01-29:
Thanks for the report.
This definition for M18/M84 is from the MarlinFW site.
_"If a timeout is given with S, this command just sets the stepper inactivity timeout."_
With your firmware M18 S0 might leave the steppers enabled forever.  In other firmware it could immediately disable the steppers because the timeout was set to "0".  There also appears to be a lot of confusion among users as to what the command actually does.  Not all users are savvy about gcode.
The max disarm timeout on my Ender is 14400 seconds (4 hours) and M18 S0 does leave the steppers enabled on my printer with Marlin 1.1.8.  I'm pretty sure that wouldn't always be the case.

A workaround for you would be to enter M18 S0 in the "gcode before pause" box.  Pause at Height will be slightly different in Cura 5.3 and the "gcode before" and "gcode after boxes" will accept multi-line commands and so this issue will have a solution in Cura.  You will be able to enter "M300,M18 S0" (comma delimited) and the commands will show up in the gcode.

So I'm going to remove the bug label and close this.  It's expected behavior and is really a "firmware dependent" thing so for safety sake it's the way it is.  Additionally there is a viable workaround in place.  I hope you understand.

closed 2023-01-29 by GregValiant

engengr, 2023-01-29:
Thanks for the reply on this.  Maybe consider tweaking the on-screen explanation to explain better what happens when leaving zero in this field means the steppers will timeout according to the firmware default setting.  I ruined at least 6 two-hour prints where the resume didn’t work and the print failed before I figured it out.  Below is the current explanation.  I think this implies that if you set this to 0 that no duration will be set thus the steppers won’t time out.  But in my case, the firmware default still applies so they timed out in 120 seconds.  Maybe change explanation to say, “Set this to 0 if you don’t want to set any duration and let steppers disarm according to firmware default.”  And maybe add, “Caution – If steppers disarm before selecting resume, the extruder may not return to the proper position.”  Or something like that.  

 

 

Hey if you choos not to, no sweat.  I have it figured out now.  But maybe could help someone else out.  I was wondering if you could add multiple lines of G-code with comma separation.  Thanks for the tip.

 

Have a great day.

 

Jay Duke

724-454-6187

GregValiant, 2023-01-29:
This is the new description from the Cura 5.3.
"After this time steppers are going to disarm (meaning that they will lose their positions). Set this to 0 if you don't want to set any duration and disarm immediately".
That as it appears to be correct per the MarlinFW definition.  It may have to be revisited as different firmware flavors can apparently be configured differently.  As I say - a lot of confusion.

engengr, 2023-01-30:
So will setting this to zero in the new 5.3 version cause the steppers to disable immediately?  Or will it cause them to disable per the default configuration set in Marlin?

 

From the Configuration.adv.h file, I see the following:

 

 

I haven’t tested this, but I think this means that if I were to change 120 to 0 in the above config file in Marlin that the steppers would not disable ever.  So in 5.3, does it add a line of g-code M18 S0 to disable the steppers, or does it do nothing and let the printer disable per the default set in Marlin?  Or does it do something else?

 

 

 

Jay Duke

724-454-6187

GregValiant, 2023-01-30:
It's up to your particular firmware.  The question is how does any particular printer handle M18 S0?  There is no way to tell if Weedo handles it the same as Geeetech.  There are a lot of people who have compiled their own custom firmware as well.
I would continue to ignore the command if that box is set to 0.  It's the safest thing to do.  If a user wants it then the "Gcode before pause" box will accept it and properly insert it.  I just don't see a "One size fits all" here.