Ultimaker/Cura #14413: Add Checkbox to "Pause at height" Post Processing Script to provide Beep at Pause

engengr, 2023-01-29:
### Is your feature request related to a problem?

When using the Post Processing Scripts, "Pause at Height" feature, it would be nice to add a checkbox that if checked, it would insert a line of g-code such as "M300 S880 P300" so the printer will beep when it pauses.  I can do this with the "G-code Before Pause feature as a work around.  But for others, this might be a handy feature to notify them that their printer has paused.

### Describe the solution you'd like

When checking the box to add a Beep at Pause, the printer will sound a Beep tone notifying the operator that the printer has paused.

### Describe alternatives you've considered

Using the G-code Before Pause to add the code "M300 S880 P300".

### Affected users and/or printers

Users that use the Pause at Height feature and are not knowledgeable about g-code programming will benefit from the ease-of-use feature.

GregValiant, 2023-01-29:
Thanks for the suggestion.
There will be changes to the PauseAtHeight plugin in 5.3 (my modest first contribution to Cura).  This isn't one of them.
Not all firmware supports the P and S parameters.  M300 as a standalone command could work.  My Ender will accept the parameters but ignores the S parameter.

Consider - no matter what someone puts in as the P parameter some people will complain that it's too long and others will complain that it's too short.

GregValiant, 2023-07-02:
The feature has been added so I'll close this.

closed 2023-07-02 by GregValiant