Ultimaker/Cura #14415: Filament Change Script still active after removing

spazzb0t, 2023-01-29:
### Application Version

5.2.2

### Platform

Windows 11

### Printer

Anycubic Vyper

### Reproduction steps

Sliced a model with Change Filament Script added for 2 layers using Marlin (M600 U). Set settings to have 0mm retraction and 0mm unload. Script 1 change at layer 222, second script change at layer 242.

### Actual results

Changes took place but filament still retracted and unloaded. After both changes completed, filament would randomly unload. More than 30 occurrences happened  between later 242 and 575.

After print completed, sliced another file, without filament changes and again they were happening very randomly.

Re-sliced the file in Cura 5.2.0 after trying to duplicate the issue but was unsuccessful. 

### Expected results

Expected change filament script to only execute where it should. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

File uploaded is a 2 day print

GregValiant, 2023-01-30:
Thanks for the report.
You can open the gcode in a program like Notepad++.  It's search function has a "count" tool and will tell you how many times M600 appears in the file.
The Filament change plugin is pretty simple.  It looks for one layer number and puts M600 right above it.
Did you have any otherr post-processors active?

spazzb0t, 2023-01-30:
Hi,Sorry I had already deleted the file already. I thought of this after the fact as well but it was too late.Sent from my iPhone

GregValiant, 2023-01-30:
I haven't heard of this behavior before.  Enough people use the Filament Change plugin that this should have been noticed.

I'm going to remove the bug label and close this but if you re-create the problem or it occurs again save the files and let me know here and I'll re-open this and put the bug label back on.  Without files to check there is no way to investigate what happened.  I hope you understand.
(As a side note:  I'm pretty confident that there were only two M600 lines in the gcode so guessing from afar it may have been a printer/firmware problem?)

closed 2023-01-30 by GregValiant