Ultimaker/Cura #14422: Layer preview shows only 1 layer

Wistra2023, 2023-01-30:
### Application Version

Wanhao 1.1 beta

### Platform

Windows 10

### Printer

Wanhao Duplicator 6

### Reproduction steps

1. slicing the model
2. view layer

### Actual results

only top layer is visable

### Expected results

all layer under the actual should be visable

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2023-01-30:
Uncheck "nur obere schichten anzeigen".  Let us know if that works.

GregValiant, 2023-01-30:
"Only the movements are shown".  I am having trouble understanding.
I see that Cura is in "compatibility mode" and is showing the preview in flat lines rather than 3d view.  That can be caused by a couple of things.

- If your video system does not support OpenGL 4.0 or your computer has 2 video systems and Cura is using one that doesn't support OpenGL 4.0
- The "Preferences / Configure Cura / General / Viewport Behavior" setting "Force Layer View Compatibility Mode" is checked.

fieldOfView, 2023-01-30:
Also note that what you are using is based on a _very_ old version of Cura, and honestly we don't know what changes Wanhao made to that old version of Cura. We really can't help you much beyond what @GregValiant said above. Perhaps you can ask Wanhao for support instead?

closed 2023-01-30 by fieldOfView