Ultimaker/Cura #14425: slicing error

Jeffphoenixgold, 2023-01-30:
### Application Version

5.2.2

### Platform

Windows 10

### Printer

Anet ETX5

### Reproduction steps

resized file, then went to slice

### Actual results

Says "slicing Failed"

slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker

### Expected results

should have sliced file

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[image]

[image]

Jeffphoenixgold, 2023-01-30:
[image]

MariMakes, 2023-02-03:
Hey @Jeffphoenixgold ,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

This mesh seems faulty
Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools
If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/.

Is it possible that your printjob is located at 0, 0, 0, and it slices if you move it a little away from the center of the buildplate? 
Then it might be a duplicate of https://github.com/Ultimaker/Cura/issues/13643

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

Jeffphoenixgold, 2023-02-04:
Thank you for replying back. I will definitely try these ideas.

Thanks

Jeff

MariMakes, 2023-02-13:
Hey @Jeffphoenixgold,

Can you give us an update? 
Is it still an issue or can we close the ticket?

Jeffphoenixgold, 2023-02-13:
Thank you for your help. Yes you can close the ticket.

MariMakes, 2023-02-13:
Thank you! Happy printing!

closed 2023-02-13 by MariMakes