Ultimaker/Cura #14426: Cura adds unrequired support to boat benchy

kripper, 2023-01-30:
### Application Version

5.2.2

### Platform

Windows 10

### Printer

Creality Ender V2 Neo

### Reproduction steps

1. Setup printer with PETG.
2. Load the popular boat benchy
3. Slice with support enabled 

### Actual results

Cura adds unrequired support to boat benchy

### Expected results

Support should only be added where really required.
Cura knows the density and parameters and should be able to calculate correctly.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Provieded info is sufficient.

Asterchades, 2023-01-31:
The provided is very much _not_ sufficient. It's impossible to know from what you've given where this particular support is located, what settings you've used to determine the location of the automatic supports, and why you believe it to be superfluous. This is further hampered by multiple iterations and versions of the Benchy model existing - many of which are not manifold.

kripper, 2023-01-31:
@Asterchades I provided the STL file.

See more details [here](https://www.3dbenchy.com/features/).

MariMakes, 2023-02-03:
Hey @kripper,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

@Asterchades is correct. We are missing the project file. It contains the printer and settings we need for troubleshooting. Things like support placement are really dependent on your material, setting and printer choice. 

To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

kripper, 2023-02-03:
Ok. Here I generated a project file:
[CE3PRO_Boat.zip](https://github.com/Ultimaker/Cura/files/10580573/CE3PRO_Boat.zip)

MariMakes, 2023-02-10:
Hey @kripper,

Thanks for sharing your project file. 

I took a look and Cura seems to generate support as it's been instructed. If you look at the model from below you can see that the red area's indicate that it's an area that needs support. 
[image]

What you are seeing seems to be related to the Support Overhang Angle it seems to be too low for this model.
If you increase it to 80 you see that you have a lot less area that needs support. 
[image]

Resulting in the support you were probably expecting. In experience is this a model that doesn't really need support. 
[image]

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

kripper, 2023-02-10:
The thing is, it's known that this popular benchy doesn't require supports because of its small size:

"The 3D model is designed to print at 1:1 scale without support materials." [[1](https://www.3dbenchy.com/features/)]

Asterchades, 2023-02-10:
The instructions are telling you not to use supports - not that they shouldn't exist at all on the model. It has overhangs, it has bridges, so of course it's going to have positions where supports could be used, but the idea is that you don't use them in order to properly stress test your printer.

kripper, 2023-02-10:
Right. Using supports is the safe way.

The overhang angle should depend on the printer and probably also on the weight of the overhanging material. Maybe its worth taking this in consideration to avoid using supports on small prints with light overhangs. Maybe not a bug, but a feature request?

Anyway, I'm happy with Cura as is and just wanted to confirm.

MariMakes, 2023-02-13:
Hey @kripper,

The settings that are available in the Recommended mode are selected for a human to define.  
Because their values are dependent on the size, orientation, and intent of your printjob.
Enabling support is one of these settings. 

[image]

It will take a while and a lot of AI training data to have Cura make these decisions for you. 
That's a strategic decision that's outside of my control 😉 
I hope you understand.

closed 2023-02-15 by MariMakes