Ultimaker/Cura #14427: Post processing change temperature at layer puts M104 and starts to print without waiting

bellzw, 2023-01-31:
### Application Version

5.1.0

### Platform

64 bit Windows 10 v. 21H2 build 19044.2486 running on a Lenovo Yoga 14

### Printer

modified BIBO (I added overrides for actual max speeds, accelerations, etc.)

### Reproduction steps

1. Load temp tower stl (https://www.thingiverse.com/thing:2729076, the full PLA tower)
2. Extensions->Post Processing -> Modify G-Code
3. Using brim adhesion, set initial temperature to 200
4. Either 
        a. Add script multiple ChangeAtZ5.3.0 (Experimental), select Change Extruder 1 Temp and set a value and height, or
        b. Add TempFan Tower and set starting temp, layers between changes, increment, layer offset, fan tower not enabled.
5. Slice
6. Save file

### Actual results

Find M104 Sxxx inserted at the correct points (just before layers 7, 57, 107, etc.), and extruding starting well before the nozzle could have moved to the correct temperature.  At least I think that's what would happen if M104 is followed within a few instructions with extruding G1 commands.

These are the screen shots of the Post Processing data entry screens. 
[image]

[image]

This is a zip file with the gcode from slicing with ChangeAtZ5.3.0 and with TempFan Tower.
[Temperature Tower.zip](https://github.com/Ultimaker/Cura/files/10541093/Temperature.Tower.zip)

This is a screenshot showing the machine parameters I supply to Cura with my own BIBO printer definition.  Heat up speed is 2.5 C/s, cool down is 0.55.  Both were measured by me.
[image]

### Expected results

I would have thought that Cura would calculate the time to heat up and cool down (it seems to do that for prints without such temperature changes) and start those processes early enough so that extrusion of each level of the tower would be at the correct temperature, or would have inserted an M109 rather than M104.   My printer would need about 10 s to heat from 200 to 225, and then about 18 s to cool by 10 C at each tower level change. Have I misunderstood how M104 works or how Cura uses the heating and cooling rates?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I can provide my BIBO printer definition if you need that.  BTW, my .def.json file fixes an error in what is distributed with Cura.  Can I can suggest a new definition, or does that need to come from the manufacturer?

bellzw, 2023-02-02:
Edited the original to correct the order of selecting brim and starting temp, and inserting temperature changes by Modify G-Code.

MariMakes, 2023-02-03:
Hey @bellzw,

Thanks for your report 👍

Before I jump in and work on your problem, have you considered to use [AutoTowersGenerator](https://marketplace.ultimaker.com/app/cura/plugins/Kartchnb/AutoTowersGenerator) or [Calibration Shapes Plugin](https://marketplace.ultimaker.com/app/cura/plugins/5axes/CalibrationShapes) to create temperature towers? 
In my experience, they are easy and work great!

Can you help me understand your problem? Perhaps can you phrase it as:
I saw X but I expected Y, and now I can't Z.

PS. Since Cura is open source, you or anyone else is more than welcome to work on adding this 3d printer and create a pull request yourself. You can find instructions on how to add a new 3D printer definition here: https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura 
If you are stuck in any way, please reach out! I'll be happy to help out. s

bellzw, 2023-02-03:
I’m sorry that I was not clear enough.  It’s not really a “I saw X but I expected Y, and now I can't Z” problem.  The issue is basic, and perhaps it is my misunderstanding of the difference between M104 and M109 and what a printer does when it processes a command to extrude.

This is what I see:  insertion of a temperature change at a layer using Modify G-Code or AutoTowersGenerator results in the insertion of an M104, followed immediately by a few G0 commands and then G1 … Ex.xxx.  By my calculation using the distances in the G0 commands, no more than 0.5 s passes between an M104 S225 and a G1 ordering extrusion.  At this point, however, since the nozzle was at 200 C, it could not possibly have been close to 225 C, and printing would start at the wrong temperature.

What I expected: insertion of a temperature change at a layer would either insert M109 or waste enough time that the nozzle reaches temperature by the time the next G1 … Ex.xxx is executed.  

Why this might be a problem: Printers require time to change the nozzle temperature, so if the next extrusion command occurs too soon, the printer will not be printing at the desired temperature.  Consequently, parts of a temperature tower layer are not printed at the correct temperature.  My printer, for example, heats at 2.5 C/s and cools at about 0.5 C/s (and Cura knows this from the printer definition).  Consequently, when a heat-up by 5 C is required, my printer needs 2 seconds to reach temperature.  Temp towers, including the AutoTowerGenerator, start high and go low by 5 – 10 C/s per step, so cooling by even 5 C, would require my printer to wait 10 s before printing.

Do I misunderstand how M104 and M109 work?  Have I misunderstood when Cura thinks the nozzle will reach temperature?  I found no M109 codes in the g-code file, except for the ones that are in the printer’s startup code.  It seems that that Modify G-code and AutoTowersGenerator are putting the wrong codes in the file.  I've attached the g-code file (has to be txt to attach it) for a temperature tower created with AutoTowersGenerator.
[Preset Temperature Tower - PLA.txt](https://github.com/Ultimaker/Cura/files/10582327/Preset.Temperature.Tower.-.PLA.txt)

rijkvanmanen, 2023-02-13:
I get your point @bellzw. But these post processing scripts are quite simple and don't do any of the calculations that [cura engine](https://github.com/Ultimaker/CuraEngine/blob/927855d0b75441a0aa52725ba5448e388ba35073/src/LayerPlan.cpp#L1560-L1636) is doing for inserting the preheat/precool commands at the right moment.

You could of course contribute to these scripts by changing the M104 to a M109, causing it to wait on the temperature. But that might introduce oozing, so not sure if that really improves the situation.

bellzw, 2023-02-13:
Indeed, I realize that the scripts can be simple; they can be a much more convenient way of inserting specific codes into the gcode files than using a text editor, especially if codes need to be added every 10,000 lines or so!  I haven’t tried changing the M104 for an M109, yet.  I agree oozing could occur if an M109 is used, which means that a retraction would also be advisable if that change is made.  Perhaps this plug-in can make it an option (default is M104, check a box for M109)?

 

As I think about what’s actually happening, having an M104 probably results in the first layer, more or less, of a new level of a tower printed at a higher temperature than the rest of the layers of the level (temperature going down as levels go up).  Maybe that won’t affect the tower much, if at all.

tlhintoq, 2023-03-03:
More specifically... You want an M109 with the R flag.
By default M109 uses the S flag.  `M109 S225` for example.  And the S flag only holds printing if the temperature is INCREASING.  That's not the case with most temp towers.  usually they start hot and cool with each tier until failure.

So you want to change from M104 S### to M109 R###

I gave up on getting them changed.  If I I build a temp tower I just let Cura do its thing, then run the gcode through Notepad++ to do a find and replace.

bellzw, 2023-03-04:
You’re right!  I hadn’t noticed until you brought it to my attention.  What would be best, of course, would be for the post processor to realize what the previous temperature was, and insert the correct code.

 

The only reason I brought this possible issue up, was somebody in my printer user group was having problems with a temperature tower, and I don’t print towers often, and I really hadn’t thought about this until I looked to see how Cura put in the temperature changes.  That’s when I saw the M104 insertions, and realized that for my printer (a BIBO dual), at least, it would take 10 seconds for the nozzle to cool by 5 degrees, so starting the new layer without waiting would mean that maybe one or more of the lower layers of each tier would be printed at a higher temperature than the rest.  The purist in me (or maybe the obsessive in me?) wanted each tier to be printed at the correct temperature in its entirety.  Probably, it doesn’t make much difference to a tower, since I haven’t had problems with mine.  Next time I print a temperature tower, I’ll need to do the M109 experiment.

tlhintoq, 2023-03-04:
@bellzw 
> What would be best, of course, would be for the post processor to realize what the previous temperature was, and insert the correct code.

I don't think it needs to realize anything.  Just needs to use `M109 R`.  
Its a _no harm, no foul_ kind of call.  If the temp change is increasing, then it works.  If the temp change is decreasing, then it also works.  No thinking on the part of software required.

MariMakes, 2023-06-05:
Hey @bellzw,

Quick update from our side 👋

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I stumbled upon your issue about M104s. Is this still an issue for you?

Would you still like to keep this issue open or do you agree that we can close it? 
If you would like us to keep it open, what's the best way that we can help?

bellzw, 2023-06-05:
> It still inserts M104S insteads of M109R... So yeah, its still an issue.
> As mentioned earlier there's no down side to making the change.
> Sure would be a shame to not fix this because more time was spent deciding if it should be fixed than actually fixing it.

I agree.  It would remain to be seen if oozing while waiting becomes a worse problem.  Maybe the solution is to have two options for a temperature tower and let the user decide which to use?

 

Zane

MariMakes, 2023-06-06:
Hey 👋 ,

I took another look at your issue. 
I noticed that a few months we merged this PR https://github.com/Ultimaker/Cura/commit/725d1026ddeee535c4b19abd983228378fb88181, it adds an option in the post-processing script that you can pick yourself if you want M109 or M104 for a pause. 
This feature is shipped with Cura 5.3.

[image]

I'll ask the team if it's possible to add something similar to this in the ChangeAtZ plug-in. 👍

bellzw, 2023-06-06:
I’m still using 5.1.  Too lazy to update, I guess. :)  I’m actually a little surprised that I don’t get a notice that a newer version is available.  I remember getting those in the past.

 

However, adding that update to TempFanTower would be a good way to implement it.

 

Would you add the option to the various Temperature Towers available under Auto Towers?  Or just to the Custom Temp Tower?

 

Zane

 

I noticed that a few months we merged this PR  <https://github.com/Ultimaker/Cura/commit/725d1026ddeee535c4b19abd983228378fb88181> 725d102, it adds an option in the post-processing script that you can pick yourself if you want M109 or M104 for a pause.

 <https://user-images.githubusercontent.com/40423138/243645683-0205b02c-8d43-486e-99fb-b6d07aa393fe.png>

bellzw, 2023-06-07:
> Hey @bellzw <https://github.com/bellzw> ,
>
> Do you mean these AutoTowers?
>  <https://user-images.githubusercontent.com/40423138/244107276-1bd2775c-32f9-43f6-a45f-7687c83f8580.png>
>
> That is not maintained by Cura. 🤔
> It's a plug-in from the marketplace, and maintained by our amazing community member kartchnb.
> If you ask nicely they might introduce it.
> You can leave a feature request here: https://github.com/kartchnb/AutoTowersGenerator/issues
>
> I discussed if it's possible to add something related to M104 and M109 in the ChangeAtZ plug-in but we believe we will not work on this anytime soon. Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.
>
> I can even help if you are stuck and tag me in the pull request. This is the file that needs the change: https://github.com/Ultimaker/Cura/blob/main/plugins/PostProcessingPlugin/scripts/ChangeAtZ.py
>
> I'll close the ticket now, I hope you understand.

Yes, I did mean these other temp towers.  I’ll contact kartchnb to see if there’s any interest.

 

If I were a better programmer, I might take you up on the offer to work on this.  

 

Zane

tlhintoq, 2023-06-10:
If anyone wants to upvote this on the AutoTower github by `kartchnb` here a new issue I just opened over there for it.
https://github.com/kartchnb/AutoTowersGenerator/issues/75