Ultimaker/Cura #14437: Load proper gcode from a .gcode file

SLiCK-303, 2023-01-31:
### Is your feature request related to a problem?

I dont like that you can create a .gcode file with this program, and then reload that same file, and the gcode will not be loaded.

### Describe the solution you'd like

if you load a .gcode file, it should have the gcode in the Extensions->Post Processsing->Modify G-Code

### Describe alternatives you've considered

clueless

### Affected users and/or printers

everyone using this program

SLiCK-303, 2023-01-31:
In addition I dont understand why when you do add gcode to a file via Extensions->Post Processsing->Modify G-Code, slice the file, save it to a .gcode file.  Then when you reload Cura, and even tell it to load a new project, that gcode is still in there, why?  It should come up with no gcode in Extensions->Post Processsing->Modify G-Code, unless you load a .gcode file.

GregValiant, 2023-02-02:
I'm not really sure what it is you are looking for.

A gcode file is a simple text file that contains the machine instructions for the printer.
The tools in "Post-Processing / Modify Gcode" are python "scripts" meant to change the gcode generated by Cura after it leaves the Cura Engine, but before it is written into a gcode file.  Once it is in a gcode file then Cura considers it carved in stone.

If you have Post-Processors Scripts active then they are attached to the "printer" by Cura.  If that printer is active then the Post-Processor box will be populated by the post-processors scripts you enabled.  The only time that is not true is when the Cura build plate is empty.

If you open a "Gcode" file in Cura, you will get the preview.  Within Cura there is no way to alter an existing gcode file.  You would need to use a text editor program and change the file manually.

I have two post-processors that are active at all times.  It would annoy me greatly if they disappeared every time I started Cura.

If you want to clear a post-processor from the list you would need to open the Post-Process dialog and click on the "x" button to the right of the post-processor name.