Ultimaker/Cura #14438: Error running cura from source

gconcha1, 2023-01-31:
### Application Version

5.2.2

### Platform

Windows 10

### Printer

3D Blocks One MkII

### Reproduction steps

I was following the guide below, and in step 3, when I tried running "conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv" the following error was presented. How can I solve this issue? 

https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source

PS: Sorry if this was asked before, but I couldn't find anything

### Actual results

[image]

### Expected results

The initialization of the Virtual Python Development Environment.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://github.com/Ultimaker/Cura

MariMakes, 2023-02-03:
Hey @gconcha1,

Can you confirm if this is the same issue as https://github.com/Ultimaker/Cura/issues/14337#issuecomment-1413579618?
Or do you need one of our developers to take a look?

gconcha1, 2023-02-03:
Hi! 

Yeah it is, I was desperade at the time xD.

I'll close it.

Thanks again.

closed 2023-02-03 by gconcha1