Ultimaker/Cura #14440: adaptative layer height

Malibu948975, 2023-01-31:
### Application Version

5.2.2

### Platform

win10 64bits

### Printer

Geeetech I3 pro x

### Reproduction steps

3 model like a 4.1 lego brick home made, adaptive layer, and

### Actual results

 index of middle variance offset with the first layer not the the "all layer height" setting. To have the result i should put 1rst layer height as tall that all layers not wanted.

### Expected results

the middle of variance between thiner and taller layer must index on general layer height settings not the height of first layer, it would be great to put on option on seize of layer view to dont care size of first layer to see ion great color dispertion the variance of significative layers except the first that is set fix.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

GregValiant, 2023-01-31:
Thanks for the report.
Look closely at the Settings Box in your screen shot.  You see you have the top setting "Layer Height" set to 0.29 and the "Initial Layer Height" set to .145.  I think you have them backwards and Cura was doing what it was told.
I don't see a bug here but rather a settings issue.  I'll leave this open so you can respond but I will remove the bug label.  Let me know if you feel that is premature.

GregValiant, 2023-02-06:
I'll go ahead and close this then.

closed 2023-02-06 by GregValiant