Ultimaker/Cura #14445: Cura 5.X.X does not Start on Windows 10 the last Version is 5.0.0 stable

Hates1209, 2023-01-31:
### Application Version

5.X.X

### Platform

Windows 10 64 Bit

### Printer

Not the Problem

### Reproduction steps

I try to install all version up to 5.3 and the last one which can I open after installation is 5.0.0 =(

I Restart everthing, deinstall the complett Windows and start with an new one fresh installation.

But nothing is working, I installed all Versions.

The installation is finished fine and after that it is not starting. (Antivirus is deactivated)

### Actual results

The installation is finished fine and after that it is not starting. (Antivirus is deactivated)

### Expected results

Nothing

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2023-02-01:
Thanks for the report.
@Hates1209 can you tell us more about the computer (processor, video sub-system, memory, etc).  There are a lot of people running those Cura versions on Win10 (including me) who haven't had this problem.  Since the problem is keeping you from providing a log file then knowing more about the computer may help.

Hates1209, 2023-02-01:
Hi,

It is an Acer Alpha Switch 12 with the core i7 6500u and 8 GB Ram. They has an SSD and all other Version also the 5.0.0 are working fine!

MariMakes, 2023-02-03:
Hey @Hates1209,

Welcome to the Ultimaker Cura Github 🚀 
Could it be related to a multi-monitor set-up? 

I have a couple of questions to narrow it down.
	1. Do you have multiple monitors? Does Cura start if you disable one of the monitors?
	2. Do you have two monitors placed above each other? Can you try this workaround from the community? https://community.ultimaker.com/topic/42633-cura-522-stopped-working/#comment-318105
	3. Do you have GPU Tweak OSD overlay enabled? Does Cura start if you disable that software?
	4. Do you have a Ryzen Graphics card? Can you try this workaround? https://support.makerbot.com/s/article/1667412437627
	5. Do you have the Nahimic audio enhancement service enabled? Can you try this workaround? https://github.com/Ultimaker/Cura/issues/12102#issuecomment-1157907665
	6. If your problem is related to scaling, can you try this workaround from the community? https://community.ultimaker.com/topic/41458-ultimaker-cura-510-window-scaling-problem/#comment-311838

Hates1209, 2023-02-05:
Hi thanks for the reply!

1. Yes I have two. If I diable one or both and use only the screen from the Laptop it is the same problem.
2.  Not my problem
3. No it ist an Intel driver and has no option for OSD
4. no grphic cards 
5.  - 6. the same nothin of my problem

When I open the exe File for Cura 5.X.X not 5.0.0 than disapperas nothing on the display. No error messages and nothing as nothing. I try all versions up from 5.0.0 an every has the same effected. Only the 5.0.0 run prefect.

closed 2023-03-03 by github-actions[bot]