Ultimaker/Cura #14448: Slicing Stops @ ~10%

Karithay, 2023-02-01:
### Application Version

5.2.2, 5.2.1, 5.0

### Platform

Windows 11 / AMD Ryzen 7 5800 8-Core Processor 3.40 GHz / 16.0 GB RAM / AMD RX 6800 XT16GB GDDR6

### Printer

Creality CR-6 SE & Mingda Magician X

### Reproduction steps

Loaded the file AEPWAR01 - 4x4 Chinampas Half (Empty).stl for slicing.

Hit the Slice button

### Actual results

The slicing progress gets to ~10%
[image]

Then slicing stops and it throws up an error box, asking me to click the link to report the bug
[image]

This is not the only file in this set that this error occurred with, but some of the files, even those with plants,  did slice (and print) normally)

This was attempted in 5.2.2, 5.2.1, 5.0, and 4.1 for both my Creality CR6-SE and my Mingda Magician X. It only actually sliced the file in Cura 4.1

### Expected results

It should have sliced normally so that I could print the file.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

5.2 Log [cura5.2.log](https://github.com/Ultimaker/Cura/files/10553345/cura5.2.log)
5.0 Log [cura5.0.log](https://github.com/Ultimaker/Cura/files/10553346/cura5.0.log)
4.1 Log [cura.log](https://github.com/Ultimaker/Cura/files/10553348/cura.log)
Project file [AEPWAR01 - 4x4 Chinampas Half (Empty).zip](https://github.com/Ultimaker/Cura/files/10553354/AEPWAR01.-.4x4.Chinampas.Half.Empty.zip)

This bug is similar to that in https://github.com/Ultimaker/Cura/issues/14441, as it mentions ending at ~10%

GregValiant, 2023-02-01:
Thanks for the report and the files.
When I zoom in on the model in Cura I get this:
[image]

So although the model is "watertight" there appears to be something going on internally.  The repair utilities I use didn't really note any errors though.  The model should have sliced.

There is an often reported bug in Cura 5.x that struggles with models like this that have "dirt" or "tile" floors.  There are a LOT of non-planar surfaces all very close to being on the same plane.  A fix for this was added into Cura a few days ago to address the bug and 5.3 will be available shortly with the fix.  Hopefully it works.

A workaround for the bug was to move and/or rotate the model on the Cura build plate.  This wouldn't slice for me when I opened the project, but did after I moved the model a ways towards the left front.
[image]

I'll mark this as a duplicate "slicing error" and close it as the Cura team has a LOT of evidence of what goes on there and it has been addressed.  I hope you understand.

closed 2023-02-01 by GregValiant

Karithay, 2023-02-02:
Thank you for the prompt feedback. Kicking it into a corner and rotating it did allow it to slice.