Ultimaker/Cura #14449: printing iniutial layer with M109 S0????

redzabu, 2023-02-01:
### Application Version

5.2.2

### Platform

Linux

### Printer

Ender 3 S1 (PRO)

### Reproduction steps

Print Settings / Material / Printing Temperature Initial Layer

Info Box: "Set at 0 to disable special handling of initial layer"

set at 0

### Actual results

M104 S0 ; Start heating up the nozzle                                                                                               
...                                                                                                 
M109 S0 ; Finish heating the nozzle ...                                                                                             

### Expected results

M104 S200 ; Start heating up the nozzle                                                                                               
...                                                                                                 
M109 S200 ; Finish heating the nozzle ...                                                                                             

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Nothing more to say

GregValiant, 2023-02-02:
Thanks for the report.
The Creality Ender 3 definition file was submitted by either Creality or a community member.  Included within that definition is the StartUp G-Code.  Within it there are the lines:
M104 S{material_print_temperature_layer_0} and
M109 S{material_print_temperature_layer_0}
so the StartUp is explicitly asking for the "Initial Layer Print Temperature" which you have set to 0 in response to the Help Balloon.  (In this case it appears to be an "Un-helpful balloon".)

I'd say that the error is the incorrect information within the balloon.  A lot of different printers use that line in their StartUps and there is a definite contradiction between what the balloon says and what actually happens.  If the {material_print_temperature_layer_0} keyword is not in the startup then everything works as it should.  I would think that just striking that sentence from the Help Balloon would go a long way towards fixing this.

@MariMakes from the Cura team will take a look at this.  The team is scrambling as the release of 5.3 is coming up and I don't think there is time for a pull request.

redzabu, 2023-02-02:
This solves the problem for me.
Didn't understand what was meant by "special handling of initial layer" besides the temperature setting anyhow...
Tnx!

GregValiant, 2023-02-02:
@redzabu I'll re-open this so the Cura team sees it.  It appears to be an easy fix and that line does have incorrect information.

saumyaj3, 2023-09-25:
Hey @redzabu! This issue will be solved in 5.5. I'll close this ticket for now.

closed 2023-09-25 by saumyaj3