Ultimaker/Cura #14451: Lines created in spiralize outer contour mode where they shouldn't be

crisscrosss, 2023-02-01:
### Application Version

5.2.1

### Platform

MacOS

### Printer

Ender 3 V2

### Reproduction steps

1. Imported this file into Cura https://www.thingiverse.com/thing:1356547/files (pyramid5.stl)
2. Sliced with spiralize outer contour mode on and 0% infill
3. Looked through the layer previews and found these lines where they shouldn't be.
[image]
[image]

### Actual results

Layer lines are found on the preview that aren't actually found on the original model.

### Expected results

Layer lines generated should match the original model. This model was designed to be printed in Vase mode (Spiralize outer contour)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

I've added a project  file and a cura log file. Not sure if the log file will help, but I added it anyway. 
[cura.log](https://github.com/Ultimaker/Cura/files/10560838/cura.log)
[CE3E3V2_pyramid5.3mf.zip](https://github.com/Ultimaker/Cura/files/10560840/CE3E3V2_pyramid5.3mf.zip)

GregValiant, 2023-02-02:
Thanks for the report and the files.
That's an interesting model and I know it was a lot of work.
There is a tiny error at layer 131 when that set of shapes gets to full size.  I'm not sure exactly what it is but there are 4 "islands" that develop on that layer and so Cura can't spiralize it.  You can see what happens when watching the preview.  One island is printed, then the nozzle returns to the start corner before traveling to the second island, etc. until the 4 discontiguous areas are finished.  After that it's all good.
Extrusions are constant when using spiralize and so the travel moves end up being full extrusions rather than regular strings.

BUT there is a fix.  Set the "Horizontal Expansion" to 0.01 (we're talking "tiny" error here) and the problem goes away as just that little bit of expansion allows Cura to connect the areas and spiralize isn't interrupted.

I'm inclined to remove the bug label from this as it's a model problem and there is a fix for it in place in Cura.  I'll leave it on for now and leave this open so you can respond here.

Here is layer 131 with Horizontal Expansion at 0.01.
[image]

crisscrosss, 2023-02-02:
Hi there,

Thanks for your response. I understand now that it is an issue with the model, and not with cura. Setting the horizontal expansion to 0.01 seems to fix it for me too.

I'll go ahead and close this issue now as the problem has been resolved

Thanks

closed 2023-02-02 by crisscrosss