Ultimaker/Cura #14454: AppImage Cura 5.x doesn't respect system font settings

tdeck, 2023-02-02:
### Application Version

5.2.2

### Platform

Ubuntu 22.04.1 LTS

### Printer

n/a

### Reproduction steps

I upgraded to the AppImage version from the PPA version of Cura on the same machine, going from 4.13.0 to 5.2.2

### Actual results

Something about either the new version of Cura or the AppImage install doesn't seem to be recognizing font settings, or perhaps DPI settings?

[image]

Here is a screenshot showing the contrast with two versions of Cura running simultaneously. In the background is Cura 4.13.0 from the PPA. It's respecting the system font size settings. In the foreground is the Cura 5.2.2 AppImage.

I have low vision and an ordinary DPI monitor (1920x1080), so I've set up my font settings to make my machine easy to use. That's unfortunately a bit of an accessibility issue for me because there doesn't seem to be any way to adjust the font / DPI in the Cura settings either.

### Expected results

At least menu bar and UI element font size should be the same between versions

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I'm running Ubuntu 22.04.1 LTS with LXDE.

I found some other bugs that could be the same issue, like [this one](https://github.com/Ultimaker/Cura/issues/11913), but thought it was worth filing one with a screenshot to directly compare.

MariMakes, 2023-02-10:
Hey @tdeck,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I run on windows and font size is often linked to display scale. 
We also have this workaround in the community to rescale the font, but I'm not sure it will work for you. 
https://community.ultimaker.com/topic/41458-ultimaker-cura-510-window-scaling-problem/#comment-311838

I'll bring it up to the team to see what they can do to improve it. Fingers crossed 🤞

MariMakes, 2023-02-17:
Hey @tdeck,

One of our Developers runs into this as well on a regular basis. 
We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10309

Thanks for the report! 👍