Ultimaker/Cura #14457: Help with UltiMaker Cura 5.2.2

SergoMArtys, 2023-02-02:
### Application Version

5.2.2

### Platform

Windows 8.1

### Printer

flyingbear ghost 5

### Reproduction steps

Good afternoon. Help with UltiMaker Cura 5.2.2
The program was downloaded from the official site. The program has been installed. When starting the program, it immediately gives an error.
I am attaching a photo of the error.
operating system Windows 8.1 64-bit
[image]

### Actual results

When starting the program, it immediately gives an error.

### Expected results

program launch

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fieldOfView, 2023-02-02:
What does it say after "DLL load failed while importing QTWidgets? Can you translate that back to english?

kliment, 2023-02-03:
It's the equivalent of "ImportError: DLL load failed: The specified module could not be found"

fieldOfView, 2023-02-03:
Thanks @kliment. 

I am afraid Ultimaker Cura no longer supports Windows 8.1:
https://support.makerbot.com/s/article/1667337917781
The biggest reason for that is the the Qt GUI framework Cura uses no longer supports Windows 8.1:
https://doc.qt.io/qt-6/windows.html

fieldOfView, 2023-02-03:
I'm closing this issue, because though it is regrettable that Cura no longer works on Windows 8.1, it is (apparently) a known issue and a deliberate choice.

closed 2023-02-03 by fieldOfView