Ultimaker/Cura #14461: Slicing and UI problems on RRF?

equipment66, 2023-02-03:
### Application Version

5.2.1

### Platform

Win 10

### Printer

FB Ghost5 + RRF on Mellow Fly Gemini v2

### Reproduction steps

Hi.
I have replaced motherboard on my printer and installed Duet RRF.

### Actual results

I start seeing following issues after that:
1) User interface changes. Mostly likely, it happens after first upload to printer, but that's not for sure. The most visible issue is extremely large arrows. Sometimes other small issues, what makes app unusable.
2) random slicing or data transfer problems, e.g. sometimes print is crashing because strange head movement.

slicing the same model with Prusa doesn't produce any problems.

I have re-installed Cura, but it doesn't help. 

[image]
[cura.log](https://github.com/Ultimaker/Cura/files/10577513/cura.log)

### Expected results

Actually, I just like name the problem 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

I have used Cura with RRF on other boards before without any problems. 
the difference is Duet RRF configured and Gemini board.

equipment66, 2023-02-17:
Problem solved disabling RepReap integration :). 
Should be good if somebody can fix plugin or this bug :)

MariMakes, 2023-02-21:
Hey @equipment66,

Thanks for the update. 👍 
The integration is maintained by one of our community members on this Github. 
https://github.com/Kriechi/Cura-DuetRRFPlugin/
Don't forget to update them if you want it to be fixed.  

Since it doesn't seem to be a Cura issue I'll be closing this ticket.

closed 2023-02-21 by MariMakes