Ultimaker/Cura #14462: Unknown Slicing Error

SaschaUncia, 2023-02-03:
### Application Version

5.2

### Platform

Win 10

### Printer

Anycubic Mega

### Reproduction steps

Import the file and slice with PLA+ profile

### Actual results

Actual - Cura hangs then gives an 'Unknown Slicing Error'

### Expected results

Expected - Properly sliced output and preview

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

The attached model generates an unknown slicing error. I'm not sure what's going on. Lychee Slicer and Idea Maker have no issues. Surface analysis in Alibre CAD shows no issues, no non-manifolds when checking the .stl in Blender. It's a pretty simple model too.

[cura.log](https://github.com/Ultimaker/Cura/files/10577615/cura.log)

[tray thin.zip](https://github.com/Ultimaker/Cura/files/10577617/tray.thin.zip)

It's a simple tray for oil and spices in my kitchen
Tested in Cura 5.1 and 5.2

SaschaUncia, 2023-02-03:
I finally got it to slice by running it through a fairly aggressive weld operation in Blender.

Still, it threw no geometry errors when analyzed and sliced fine in other slicers. Hopefully it can still serve some use as a test model for finding the cause of the slicing failure.

MariMakes, 2023-02-10:
Hey @SaschaUncia,

Thanks for your report, that's a nice tray! 👍
Sorry, it took us a while to get back to you 😞 

We've been seeing more errors when a printjob is located at 0, 0, 0, and it slices if you move it a little away from the center of the buildplate. A significant part is resolved in https://github.com/Ultimaker/Cura/issues/13643 and I was able to slice your printjob with a version of Cura with this fix. 

[image]

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
Would you still like to keep this issue open or do you agree that we can close it?

SaschaUncia, 2023-02-11:
@MariMakes  

Sure, if the root cause has been found then I'll mark the ticket as closed.

Here is a copy of both trays, feel free print some if you have a use for them :3

[trays.zip](https://github.com/Ultimaker/Cura/files/10713091/trays.zip)

closed 2023-02-11 by SaschaUncia