Ultimaker/Cura #14464: Memory leak while monitoring print via Octoprint

Dr-Gozmon, 2023-02-03:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ender 3 Pro or Ender 6

### Reproduction steps

Started a print on two printers using Octoprint. (this may not be necessary)
Closed Cura
Re-opened Cura
Went to Monitor tab
Switch to other printer (it doesn't matter which of the two is selected to begin with)

### Actual results

Memory usage starts to increase by about 5MB per second.
Today it got to 12GB of memory used before I realized and closed Cura. 

### Expected results

Memory usage should not increase to such a level

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Including cura.log from the most recent time I tested this bug. I only let it get to about 500MB before stopping. Also including my full cura.log.1 (renamed to cura.log.1.log because github) file which has a couple more instances of this bug happening. The easiest one to find would have started at about 9:42am on 2/3/23.
[cura.log.1.log](https://github.com/Ultimaker/Cura/files/10580142/cura.log.1.log)
[cura.log](https://github.com/Ultimaker/Cura/files/10580143/cura.log)

Dr-Gozmon, 2023-02-03:
Here's a screen recording of the issue happening and the log to go with it. I didn't capture it very well but the memory usage bounces around just under 400MB normally. It only starts increasing when I switch printers. 

https://user-images.githubusercontent.com/123280196/216641398-e194c9eb-d441-433a-ad9c-6bedcc1d0bef.mp4
[cura.log](https://github.com/Ultimaker/Cura/files/10580253/cura.log)

fieldOfView, 2023-02-03:
Thanks for reporting. This should probably be fixed in the OctoPrint Connection plugin. Could you open an issue over at https://github.com/fieldofview/Cura-OctoPrintPlugin/issues instead? Copy/pasting what you entered here will do.

closed 2023-02-03 by fieldOfView