Ultimaker/Cura #14467: settings auto deselect while typing in slicer

chriswhit123, 2023-02-03:
### Application Version

5.1.2

### Platform

windows 11

### Printer

Ender 5 pro, Ender 3 pro 

### Reproduction steps

fresh install with many market place items. I use advanced mode and when I change the infill or any box it deselects after you delete so you have to reselect the box then say I want 60 percent infill after I type the 6 the box deselects again and I have to reselect it to type the 0. It causes a lot of wasted filament where this version is new to me. it does this for print temps or anything I can change

### Actual results

all the boxes with temps and infill keep deselecting making you think you set temps or flow right and infill but half what you type isn't there

### Expected results

should just be able to select a slicer setting and have it stay selected to finish your settings

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

its just cura in general doesnt matter if no files are selected or if i loaded an stl or not

chriswhit123, 2023-02-03:
[cura.log.zip](https://github.com/Ultimaker/Cura/files/10581648/cura.log.zip)

cgolchert, 2023-02-21:
Sounds the same as #14350