Ultimaker/Cura #14468: Missing "Longer LK4 X" support

BradleyFord, 2023-02-03:
### Is your feature request related to a problem?

Missing printer definition

### Describe the solution you'd like

Addition of printer

### Describe alternatives you've considered

relegated to custom config in Cura 4

### Affected users and/or printers

Longer LK4 X

BradleyFord, 2023-02-03:
@LONGER3D

GregValiant, 2023-02-03:
The Longer company provides the definition files for their printers.  There has been some activity (nozzles and such) since the middle of December.  Can you install an LK4 or 4 pro and make adjustments?  There really isn't very much in the Longer Lk 4 and 5 and Pro definitions except for the machine width, depth and height.  Those are adjustable in the Cura Machine Settings dialog.

BradleyFord, 2023-02-03:
I did try some things that I found on the internet to get Cura 5 working ok, but it didnt seem to work very well. At the moment I am just using Cura 4 because it just works even if it is not as good.
Is their anything I can do to get it added into the next release easily?

GregValiant, 2023-02-03:
I think the release of 5.3 stable is close and I have not seen a pull request to add the "X" model.
Looking at the LK4 and the LK4-X I don't see a lot of difference as far as Cura would be concerned.  There are no settings in Cura specific to a direct drive as opposed to a bowden printer.  The one thing you would want to tweak would be the retraction distance as it is usually a lot shorter for a DD as opposed to a bowden.

BradleyFord, 2023-02-04:
When installing Cura 4, Longer3D give a package of files to copy into the folders to add the LK4 X.

The changes are very minor over the LK4; mainly adding some start GCODE for the BL-Touch, z_seam_type, and also as you mention retraction.
Other than that it is the same.

I am going to move forward with a pull request for it, and also happy to support the testing.

BradleyFord, 2023-02-04:
https://github.com/Ultimaker/Cura/pull/14479