Ultimaker/Cura #14471: Idea for new slicing method

3dprinting101, 2023-02-03:
### Is your feature request related to a problem?

When doing large 3d prints, infill can take up to 25-30% of the print time only at 15% infill!

### Describe the solution you'd like

I believe this idea could cut down print times on large models. I was thinking that there could be an option in the settings that makes the layer height of the infill double the wall layer height so that for every other layer, the infill would be printed, cutting the infill time in half. For example, If the layer height was set to 0.1mm, the infill would happen every 0.2mm, etc. I think that this would be a pretty nice feature to have that no other slicers have that can cut down print time without losing exterior quality.

### Describe alternatives you've considered

I haven't really considered any alternatives, but I will update this if I think of any. :)

### Affected users and/or printers

I think everyone using an FDM 3d printer would benefit from this, especially people running 3d print farms or owning larger 3d printers.

### Additional information & file uploads

Maybe there could be an option only in advanced settings for an adjustable infill layer height for specific needs. (To keep beginners from messing up their print)

GregValiant, 2023-02-04:
The setting is in the "Infill" settings and is "Infill Layer Thickness".  You can set it to multiples of your layer height.  I usually leave it as 2X layer height.  Yes, you do need to use the Advance settings to make it available.
Because the mm³/sec flow through the nozzle can be double (or triple) the regular flow, the "experimental" setting "Flow Rate Compensation Factor" can slow the print speed down to make up for that high flow rate.  That could cause a net gain of "zero" on the print time.  If you set the Compensation Factor to "0" then the speed will stay up.
There is a similar setting in the Support section for the "Support Infill Layer Thickness".  I usually run that at 2X as well.

I'm inclined to close this as the settings already exist.  Take a look at them and see if they are what you were looking for.

3dprinting101, 2023-02-04:
Ahhh! So sorry! Thanks though!

GregValiant, 2023-02-04:
I'll go ahead and close this.

closed 2023-02-04 by GregValiant