Ultimaker/Cura #14473: Bridge walls

CDarlston, 2023-02-04:
### Application Version

5.1

### Platform

Windows 10

### Printer

Sidewinder X1

### Reproduction steps

Enabled Bridging

### Actual results

layer 18 - gcode was trying to print in thin air, I looked to find it was being caused by the wall ordering set to inside to out which I have set for better overhangs

### Expected results

bridging walls would start outside to inside no matter what wall ordering settings are

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[ASX1_storage bin.zip](https://github.com/Ultimaker/Cura/files/10608202/ASX1_storage.bin.zip)

jayesbe, 2023-02-04:
Was dealing with this all day yesterday and only this morning realized the bridge walls are being printed inside to outside. Printing outside to inside immediately resolved bridge drooping causing by bridge walls being printed inside to outside. Bridge walls should be printed outside to inside

MariMakes, 2023-02-10:
Hey @CDarlston,

Thanks for your report 👍
That's a very thin wall to bridge on 😮 

Personally I would change your Wall Line Count and your Extra Skin Wall Count to make the bridge work:
[image]

I like @jayesbe suggestion to alter the wall ordering when it comes to bridging. That would have helped a lot as well in this case. 

I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. Fingers crossed 🤞