Ultimaker/Cura #14476: 5.3.-alpha doesn't slice file properly.

N6REJ, 2023-02-04:
### Application Version

5.3.0

### Platform

win 10x64 pro

### Printer

Aquila S2

### Reproduction steps

Load file
examine to see that holes and other parts are showing
slice
observe most parts are now missing.

### Actual results

doesn't include many parts

### Expected results

slices all parts properly.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Holder SocketS.zip](https://github.com/Ultimaker/Cura/files/10608503/Holder.SocketS.zip)

Works fine in cura 4 and 5.2.1

"Finished" print from 5.3.0.-alpha
[image]
walls are stringly and not fused.
I've done days and days worth of printing from 5.3.0-alpha without issues so its not a config issue as far as I know.

GregValiant, 2023-02-04:
Thanks for the report.
Load the model and use the "File | Save Project" command to create a 3mf project file.  Zip the file together with one of those "bad" gcode files.  Post the zip folder here.

Have you tried printing one of the older gcodes that you are confident print correctly?
When I load the STL the slice looks good and I'm confident it would print properly on my Ender.

Looking at the image you posted - the skirt looks very odd.  It should not have that scalloped inside edge.  That has me thinking you may have developed a problem within the printer.
Additionally, the image doesn't really look like the STL you posted.  We always need to talk "apples and apples" here.
[image]

N6REJ, 2023-02-04:
The image I posted is exactly my point.  It doesn't create all the g-code needed to do it properly.
This what the finished product looks like when sliced with 5.2.1
[image]

This is as shown in 5.2.1-alpha
[image]
[HolderSocketS.zip](https://github.com/Ultimaker/Cura/files/10609848/HolderSocketS.zip)
[HolderSocketS_2.zip](https://github.com/Ultimaker/Cura/files/10609849/HolderSocketS_2.zip)
Had to use 2 zips cause its too big for git otherwise.

GregValiant, 2023-02-05:
Turn "Spiralize" off and make sure that "remove all holes" gets turned off as well.  They are linked in 5.2.1.

N6REJ, 2023-02-05:
spiralize did it... ty.

closed 2023-02-05 by N6REJ