Ultimaker/Cura #14478: How to update Cura manually ? New version not detected

titou4307, 2023-02-04:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

FL SUN400

### Reproduction steps

Version actual 5.2.1

 Version disponible en ligne 5.2.2

Mise à niveau de version (menu.....)
Ne trouves pas la nouvelle version....

### Actual results

Not find the new version

### Expected results

New version 
How to update ?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

How to update the cura software ?

fieldOfView, 2023-02-05:
Download the release, either via the Ultimaker website or via https://github.com/Ultimaker/Cura/releases.

titou4307, 2023-02-05:
I have downloaded Cura 5.2.2 but when I install it create a new folder with install cura 5.2.2 and not scratching 5.2.1

So does I need to install 5.2.2 and erase (uninstall) 5.2.1 ? So I will lose the parameter of my printer FLSUN ??

fieldOfView, 2023-02-06:
You can uninstall previous versions before installing a new version, but you can (normally) install versions side-by-side. Note however that Cura 5.2.2 is a "point release", so it will install in the same folder as Cura 5.2.1 by default. Cura 5.3 will install in a new folder. It should be fine to install Cura 5.2.2 over Cura 5.2.1.

If you uninstall, make sure you read the dialogs, and do not let the uninstaller remove your settings if you want to keep your current printer settings.

fieldOfView, 2023-02-06:
Sorry, I checked and I was wrong. Cura 5.2.2 will install in a folder separate from Cura 5.2.1 by default, and you can run them side by side.

titou4307, 2023-02-06:
[image]

titou4307, 2023-02-06:
You know the soft ? here I have errors.....

MariMakes, 2023-06-05:
Hey @titou4307,

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I stumbled upon your issue about updating Cura. 

Your last question was
_Does this version auto-install the keep the personalization? Or does it need to reinstall all profiles?_
I think the answer is that you will get a pop-up when a new version of Cura is available. The pop-up will include a link to where to download the new version. Your personalized files will be upgraded to the newest version of Cura. What you encountered with your FlsunV400 was related to the printer profile being shipped with Cura 5.3 and it might be clashing with how you installed your printer before it was available in Cura. 

If you are worried about losing settings, you can make a back-up before upgrading your Cura. You can learn more about how to do that here: https://support.makerbot.com/s/article/1667412704080

Has your original issue been resolved? 
Would you still like to keep this issue open or do you agree that we can close it?

titou4307, 2023-06-05:
Hello
I have uninstall the version 5.2.1...
After I have install the version 5.2.2 and all my profiles and printer

Since, there was update (at this time this is 5.3.1) and all was OK after auto update
When I say update there is a popup for the link to download the last version and after downloading click double for install
After this all was OK

My problem is solved

MariMakes, 2023-06-06:
Hey @titou4307,

Thanks for the reply. I'll close the ticket than 👍 

You can do the same as the backup does automatically. 
It saves your configuration folder as a zip online, but you can save it locally by following these steps: 
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version

I hope this helps 😊 

Happy printing 🎉

closed 2023-06-06 by MariMakes