Ultimaker/Cura #14480: BUG! Missing most Printers

C4pp3, 2023-02-04:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Creality Ender 3 V2

### Reproduction steps

Trying to add another printer to my cura (Creality CR10 Smart Pro) I see that there is only 2 printer to pick from in the printer list under ''Creality 3D''. Also seems like only 2 or 3 printers show up under the different brand drop down menues.

### Actual results

--

### Expected results

I expect to see all or atleast 90% of market printers or atleast all Creality printers in the drop down menu to add my new printer.

NOTE! I have tried copying over the profile and files from Creality Slicer 4.8.2 without results.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[cura.log](https://github.com/Ultimaker/Cura/files/10609294/cura.log)

GregValiant, 2023-02-04:
Thanks for the report.
Do you happen to have the "Startup Optimiser" loaded (from the MarketPlace)?  One of it's functions is to shorten the list of printers so Cura can load faster.

C4pp3, 2023-02-04:
Hi, well seems like that was exactly the problem.. I feel like a fool.
However nowhere did I see that it would do that, easy fix and no bug
luckily then, thank you!

GregValiant, 2023-02-05:
It's noted on the blurb within the Marketplace where you loaded it.
I'll go ahead and close this then.

closed 2023-02-05 by GregValiant