Ultimaker/Cura #14483: Continuous Fiber Fabrication (CFF) Development - ($$$ paid development)

adamrumjahn, 2023-02-05:
### Is your feature request related to a problem?

Need the capability to do continuous carbon fiber slicing rather than modifying part as a multi-material print as a hack.

### Describe the solution you'd like

A simple add-on to the slicer capable of adding continuous fiber to specific layers, specific perimeters and ultimately a user interface for allowing orientation of fibers similar to infill (linear infill, direction/orientation). We would need to be able to input the fiber width and also control the fiber extrusion temperature and flow rate/speed. The slicer must also consider the maximum bending radius of the fiber and inject g-code for cutting the fiber. Careful consideration must be taken to ensure that existing patents are circumvented for cutting and ironing of the fiber.

The simplest method would be to replace 2nd outermost perimeter plus additional inner perimeters with continuous fiber during slicing. The more advanced method would be to develop something that allows rectilinear fiber orientation (similar to Automated fiber placement, AFP, machines do) so you can choose how to orient your fiber. Fiber reinforcement should also be able to be added by selecting faces or geometry similar to paint-on support structures and modifiers.

### Describe alternatives you've considered

Markforged's slicer and Anisoprint's slicer's are proprietary and only work on their own systems. I've considered creating geometries within the print model volume to represent the fiber volume. This can work on a case-by-case basis, but an automated solution is required.

### Affected users and/or printers

The Ultimaker community (and company) can benefit from this development, including anyone else interested in developing CFF printing technologies. There is currently a lack of development in this field, partially because there are no open-source slicers offering this capability.  

### Additional information & file uploads

I have a research project that can cover a significant part of the budget for the development costs for this. If you have the capability, interest and availability for implementing this starting 01 April 2023, please DM me or reply directly in this thread.

Santoshsomi, 2023-02-24:
@adamrumjahn , can you ping me?

adamrumjahn, 2023-02-25:
@Santoshsomi , I can't send private DM's.  Please send me an email at adam@orion-am.com.