Ultimaker/Cura #14487: Add a profile for the Ender 3 V2 Neo

MrBrain295, 2023-02-05:
### Is your feature request related to a problem?

Kind of, I have an Ender 3 V2 Neo and there is no official profile for it I followed a tutorial for a profile and it has worked fine but it would be nice to have one built in.

### Describe the solution you'd like

A profile for the Ender 3 V2 Neo.

### Describe alternatives you've considered

Making my own custom profile. It has worked but it would be easier to start out with the printer if Cura had a profile for it.

### Affected users and/or printers

Ender 3 V2 Neo users.

### Additional information & file uploads

None.

MrBrain295, 2023-02-05:
I didn't see this when I made the issue but this is fixed by #13580.

GregValiant, 2023-02-06:
It may or may not make it into the next release of Cura.  You can keep an eye on that thread and see when the definition is added.  You could then copy the file and paste it into the definitions folder on your computer.
Can I close this then? (it does seem to be getting attention in #13580 ).

MrBrain295, 2023-02-06:
Yeah

closed 2023-02-06 by GregValiant