Ultimaker/Cura #14489: Creality belt saves empty gcode

PantsBeaver, 2023-02-06:
### Application Version

Creality Belt 3.6

### Platform

Windows 10

### Printer

cr-30

### Reproduction steps

1. file > open file > select .stl
2. click on "prepare" (same function as "slice" on cura 4.12.1 that I use for my other non-belt printers)
3. save to removable drive

### Actual results

When I look at the files on the SD card, it is there, and the type says "GCODE File" but the size of the file is 0 bytes. I confirm this by opening it in notepad and it opens an empty page. I also tried saving it to my computer desktop and it does something different. It doesn't save anything there, not even a file name. 

### Expected results

I would expect that it would save a useable gcode file with containing information.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

It worked once today. I opened it up and saved a file as I usually do, printed it, and then made some changes in fusion 360. The next time I tried to save the gcode is when this started happening. It is not only that one .stl by the way. I've tried it with numerous .stl's that I have printed in the past and had the same issue. I thought maybe it was a bug so I installed Blackbelt 3.6.2 and it had the same problem.  I am sure the sd card is working because I am able to open cura 4.12.1 and save gcode files with no problem. Just to be sure I have also formatted the SD card (FAT32), restarted the program multiple times, restarted my computer multiple times, and every few attempts I re-open cura 4.12.1 and slice and save a file just to keep my head on straight. I cannot figure out what to even try next.

Asterchades, 2023-02-06:
Creality's belt slicer additions were not implemented by the Cura team - they were implemented by Creality. Any diagnostics from the Cura team would effectively require them to reverse engineer the alterations that have been made to the underlying Cura client (both the extensions and the base client), not only taking an excessive amount of time but opening up plenty of avenues for mis-diagnosis.

You'll need to contact Creality for support on their custom versions. Or identify who's code they've actually used (I doubt they did their own) and contact them, in the hopes that it has been correctly licensed.

closed 2023-02-06 by Asterchades

PantsBeaver, 2023-02-06:
Thanks for the prompt reply.

FFFresco, 2023-03-30:
I had the same issue. Solved by enabling retraction settings. I didn't found other parameters triggering this sofware behaviour.