Ultimaker/Cura #14490: Build .exe from source code

KrIsChUnn, 2023-02-06:
### Application Version

5.3.1-alpah

### Platform

Windows 11

### Printer

Ultimaker S5

### Reproduction steps

the cura can run successfully from source code. then
1.On the path of cura source code,  "pyinstaller -F cura_app.py"
2.copy the "cura_app.exe" generated in the last step in "dist" to the path of cura source.
3.run "cura_app.exe" and got failure.

### Actual results

[image]
[image]
[image]
the[image]
[image]
[image]

### Expected results

cura_app.exe run successfully

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

the log file shows :
Thread 0x0000483c (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00006484 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00006708 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00002934 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
[... 212 lines cut ...]
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00000ce8 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00004ff8 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Current thread 0x00004088 (most recent call first):
  File "cura\CrashHandler.py", line 397 in _logInfoWidget
  File "cura\CrashHandler.py", line 182 in _createDialog
  File "cura\CrashHandler.py", line 86 in __init__
  File "cura_app.py", line 177 in exceptHook

MariMakes, 2023-02-10:
Hey @KrIsChUnn,

😖 Sorry to hear you are stuck.

I'm only a fledgling member of the Build From Source Club 👑 
Before we get started, I have a few questions to narrow down the troubleshooting:

1. Can you make sure you have specifically Conan version 1.56.0?  
`pip install conan == 1.56.0 --force `
You can check the [Conan Documentation]( https://docs.conan.io/en/latest/installation.html ) for the latest version and installation instructions

2. Did you try clearing the Conan Cache? 
`conan remove "*" -f `

3. Can you make sure you have the latest config with this command?
`conan config install https://github.com/Ultimaker/conan-config.git`

4. Can you make sure you have a default profile?
`conan profile new default --detect --force`

5. With these steps you should have a clean slate and you can follow the steps on our wiki again. 
[You can find it here. ](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source) 

If you are stuck in any way, please reach out. We are happy to help out. 
The best way to do that is to provide us with your complete run-out log. 
Not just the part that fails, but all the lines generated after you've given your command. 
We need it see your profiles and packages. 

This is a neat trick to keep our Github issues more manageable. 

Can you help me understand. What do you want to achieve by creating a Build?
Are you intending to share it with others or are you using it for research purposes?
I would love to know more about your project, maybe we can help.

KrIsChUnn, 2023-02-10:
> Hey @KrIsChUnn,
>
> 😖 Sorry to hear you are stuck.
>
> I'm only a fledgling member of the Build From Source Club 👑 Before we get started, I have a few questions to narrow down the troubleshooting:
>
> 1. Can [...]

May be I did not make myself clear. The evnironment has been successfully build and I can run cura from source. 
My problem is how to generate an executable file (.exe) from the source code?  
I tried "pyinstaller cura_app.py" command. However, the .exe file generated can not run successfully.

MariMakes, 2023-02-13:
Hey @KrIsChUnn 

Can you help me understand, what you want to achieve by creating a .exe?
Are you intending to share it with others or are you using it for research purposes?

I would love to know more about your project, maybe we can help.

MariMakes, 2023-02-13:
Could this perhaps be what you are looking for? https://github.com/Ultimaker/Cura/issues/13568#issuecomment-1279992912

closed 2023-03-03 by github-actions[bot]