Ultimaker/Cura #14491: Slicing failing

tgcustoms, 2023-02-06:
### Application Version

5.2.1

### Platform

windows 11

### Printer

ender 3 s1

### Reproduction steps

I imported the model, scaled it up and tried to slice it to have it fail. 

### Actual results

the slicing failed on multiple models.

### Expected results

slicing should complete and ready for print. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CE3S1_Datsun 620 KingCab 1977 - Front - 1-8-r.zip](https://github.com/Ultimaker/Cura/files/10665323/CE3S1_Datsun.620.KingCab.1977.-.Front.-.1-8-r.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/10665329/cura.log)

GregValiant, 2023-02-08:
Thanks for the report and the files.
I couldn't get it to slice either.  Cura reported it was watertight but that it has both missing and extraneous surfaces and there are 376 separate meshes within the model.
MS 3D Builder said it was fine but when trying to merge the 367 pieces it locked up.
The online repair service I use said there were no errors but then timed out trying to fix the model.
Here is the view in Mesh Mixer.  All the magenta balls are (alleged) errors.  When I tried to repair it in Mesh Mixer a lot of areas simply disappeared.  That often happens where there are non-repairable errors in the model. 
[image]

Here is the X-ray view in Cura.  The white areas match up very well to the areas that Mesh Mixer said had issues.
[image]

My guess (and it is a guess) is that all those parts aren't joined correctly and that there are small gaps between the pieces.  As a consequence, Cura can't figure it out.

Someone from the Cura team will take a look.  On this particular model - trying to figure out if it's the model or a bug in Cura won't be easy.

MariMakes, 2023-02-10:
Hey @tgcustoms,

Thanks for your report 👍
Are you sure this model is made for 3D printing? It looks like a CAD model.

Have you tried a mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/?
Or the"Make Solid" feature in AutoDesk MeshMixer. https://meshmixer.com/download.html?

closed 2023-03-03 by github-actions[bot]