Ultimaker/Cura #14493: Cura 5.2.1 AppImage Stop Displaying Models

baqwas, 2023-02-07:
### Application Version

5.2.1

### Platform

Ubuntu 22.10

### Printer

Prusa i3 xl

### Reproduction steps

1. Run appimage - Ultimaker-Cura-5.2.1-linux-modern.AppImage
2. Open File(s)...
3. Select STL file

### Actual results

Nothing is displayed. This behavior is not specific to the model file chosen. It occurs with all model files - even files that have been displayed in previous sessions.

### Expected results

The model should be displayed with an object list summary including nominal dimensions.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

The recovery (i.e. workaround) is to reboot the desktop. Thereafter, Cura will perform as expected but after two or three sessions (i.e. Cura is closed and re-opened) Cura stops displaying the models.

The desktop has sufficient RAM and DASD capacities.

baqwas, 2023-02-10:
**stderr.log** below; stdout.log was empty. _Sorry_, couldn't upload zip file for some reason. Thanks for looking into the issue.

Regards.

trimesh/curvature.py:12: DeprecationWarning: Please use `coo_matrix` from the `scipy.sparse` namespace, the `scipy.sparse.coo` namespace is deprecated.
  from scipy.sparse.coo import coo_matrix
[35m2023-02-07 21:33:55,302[0m - [36mDEBUG[0m - [37m[MainThread] UM.Application.initialize [163]: Initializing Ultimaker Cura[0m
[35m2023-02-07 21:33:55,302[0m - [36mDEBUG[0m - [37m[MainThread] UM.Application.initialize [164]: App Version 5.2.1[0m
[35m2023-02-07 21:33:55,302[0m - [36mDEBUG[0m - [37m[MainThread] UM.Application.initialize [165]: Api Version 8.2.0[0m
[35m2023-02-07 21:33:55,302[0m - [36mDEBUG[0m - [37m[MainThread] UM.Application.initialize [166]: Build type None[0m
[35m2023-02-07 21:33:55,302[0m - [33mWARNING[0m - [37m[MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/lib/uranium must be a folder.[0m
[35m2023-02-07 21:33:55,302[0m - [33mWARNING[0m - [37m[MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/lib64/uranium must be a folder.[0m
[35m2023-02-07 21:33:55,303[0m - [33mWARNING[0m - [37m[MainThread] UM. cmake -D CMAKE_BUILD_TYPE=RELEASE \
 	-D CMAKE_INSTALL_PREFIX=/usr/local \
 	-D INSTALL_PYTHON_EXAMPLES=ON \
 	-D INSTALL_C_EXAMPLES=OFF \
 	-D OPENCV_ENABLE_NONFREE=ON \
 	-D WITH_CUDA=ON \
 	-D WITH_CUDNN=ON \
 	-D OPENCV_DNN_CUDA=ON \
 	-D ENABLE_FAST_MATH=1 \
 	-D CUDA_FAST_MATH=1 \
 	-D CUDA_ARCH_BIN=7.5 \
 	-D WITH_CUBLAS=1 \
 	-D OPENCV_EXTRA_MODULES_PATH=~/opencv_contrib/modules \
 	-D HAVE_opencv_python3=ON \
 	-D BUILD_EXAMPLES=ON ..PluginRegistry.addPluginLocation [202]: Plugin location /tmp/lib32/uranium must be a folder.[0m
[35m2023-02-07 21:33:55,303[0m - [33mWARNING[0m - [37m[MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/.mount_UltimaGeXjlX/plugins must be a folder.[0m
[35m2023-02-07 21:33:55,303[0m - [33mWARNING[0m - [37m[MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/Resources/uranium/plugins must be a folder.[0m
[35m2023-02-07 21:33:55,303[0m - [33mWARNING[0m - [37m[MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/Resources/cura/plugins must be a folder.[0m
[35m2023-02-07 21:33:55,303[0m - [36mDEBUG[0m - [37m[MainThread] UM.Resources.__initializeStoragePaths [481]: Initializing storage paths[0m
[35m2023-02-07 21:33:55,303[0m - [36mDEBUG[0m - [37m[MainThread] UM.Resources.__initializeStoragePaths [491]: Config storage path is /home/reza/.config/cura/5.2[0m
[35m2023-02-07 21:33:55,303[0m - [36mDEBUG[0m - [37m[MainThread] UM.Resources.__initializeStoragePaths [499]: Data storage path is /home/reza/.local/share/cura/5.2[0m
[35m2023-02-07 21:33:55,304[0m - [36mDEBUG[0m - [37m[MainThread] UM.Resources.__initializeStoragePaths [511]: Cache storage path is /home/reza/.cache/cura/5.2[0m
[35m2023-02-07 21:33:55,304[0m - [32mINFO[0m - [37m[MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: /tmp/.mount_UltimaGeXjlX/share/uranium/resources/bundled_packages/uranium.json[0m
[35m2023-02-07 21:33:55,304[0m - [32mINFO[0m - [37m[MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: /tmp/.mount_UltimaGeXjlX/share/cura/resources/bundled_packages/cura.json[0m
[35m2023-02-07 21:33:55,304[0m - [36mDEBUG[0m - [37m[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [124]: Trying OpenGL context 4.1...[0m
[35m2023-02-07 21:33:55,304[0m - [36mDEBUG[0m - [37m[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [132]: Yay, we got at least OpenGL 4.1 core: 4.6 Core profile[0m
[35m2023-02-07 21:33:55,304[0m - [36mDEBUG[0m - [37m[MainThread] UM.Qt.QtApplication.initialize [192]: Detected most suitable OpenGL context version: 4.1 Core profile[0m
[35m2023-02-07 21:33:55,304[0m - [32mINFO[0m - [37m[MainThread] UM.Qt.QtApplication.initialize [199]: Initializing job queue ...[0m
[35m2023-02-07 21:33:55,305[0m - [32mINFO[0m - [37m[MainThread] UM.Qt.QtApplication.initialize [203]: Initializing version upgrade manager ...[0m
[35m2023-02-07 21:33:55,305[0m - [32mINFO[0m - [37m[MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from /tmp/.mount_UltimaGeXjlX/share/uranium/resources/bundled_packages/uranium.json[0m
[35m2023-02-07 21:33:55,305[0m - [32mINFO[0m - [37m[MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from /tmp/.mount_UltimaGeXjlX/share/cura/resources/bundled_packages/cura.json[0m
[35m2023-02-07 21:33:55,305[0m - [32mINFO[0m - [37m[MainThread] UM.PackageManager._loadManagementData [203]: User package management file /home/reza/.local/share/cura/5.2/packages.json doesn't exist, do nothing[0m
[35m2023-02-07 21:33:55,305[0m - [32mINFO[0m - [37m[MainThread] UM.PackageManager._removeAllScheduledPackages [280]: Attempting to remove the following scheduled packages: - [0m
[35m2023-02-07 21:33:55,305[0m - [32mINFO[0m - [37m[MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [121]: Loading plugin configuration file '/home/reza/.config/cura/5.2/plugins.json'[0m
[35m2023-02-07 21:33:55,305[0m - [33mWARNING[0m - [37m[MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/lib/cura must be a folder.[0m
[... cut ...]
[35m2023-02-07 21:35:17,622[0m - [36mDEBUG[0m - [37m[MainThread] UM.Qt.QtApplication.windowClosed [508]: Shutting down cura[0m
[35m2023-02-07 21:35:17,649[0m - [36mDEBUG[0m - [37m[MainThread] CuraEngineBackend.CuraEngineBackend._terminate [360]: Attempting to kill the engine process[0m
[35m2023-02-07 21:35:17,650[0m - [36mDEBUG[0m - [37m[MainThread] CuraEngineBackend.CuraEngineBackend._terminate [366]: Killing engine process[0m
[35m2023-02-07 21:35:17,650[0m - [36mDEBUG[0m - [37m[MainThread] CuraEngineBackend.CuraEngineBackend._terminate [369]: Engine process is killed. Received return code -15[0m

baqwas, 2023-02-10:
Hello @MariMakes,

It would be foolish for me to comment on Cura internals but I don't think that this issue is related to [#1415](https://github.com/Ultimaker/Cura/issues/14145). Rationalization on my part: my issue is occurring during display. I feel that the file is read correctly but the "_display manager_" does not complete its work to render the image. As a test, I tried I a file with known errors and Cura immediately reported an error in processing that file.

Regards.

MariMakes, 2023-02-13:
Hey @baqwas,

Do I understand correctly that you think that your issue might be a duplicate of https://github.com/Ultimaker/Cura/issues/14145?

This issue specifically occurs when you click Open File. This window is not appearing:
[image]

Is that what you are encountering?

baqwas, 2023-02-13:
Hello @MariMakes,

Thanks again for following up. My issue is different from that reported in #14145.

In my case, Cura does not display the file at all after the Open command is processed. Selecting New Project and opening other valid files does not change the _status quo_ - nothing is displayed. If I do a cold reboot and repeat the steps, the files are correctly displayed.

There is nothing specific about the files that are not displayed. Any file that I choose will not be displayed. This anomaly occurs after I close Cura (1st session after logon) and start the program again. In other words, the first session of Cura after reboot works fine but subsequent sessions manifest the anomaly. This behavior is consistently repeatable. Let me know what information you need (that I can collect at my end) to help me overcome this roadblock.

Regards.

MariMakes, 2023-02-13:
Hey @baqwas,

That is indeed different. 👍 

I'll bring it up to the team to see what they can do to improve it. 
Fingers crossed 🤞

MariMakes, 2023-05-01:
Hey @baqwas,

Sorry, it took us a while to get back to you 😞 
We might have a solution for you in a nightly build that has been made available. 

You can download it here: https://github.com/Ultimaker/Cura/discussions/15098
Can you confirm if this resolves your issue? Also, let us know if this doesn't resolve your issue then we'll take another look. 😉

baqwas, 2023-05-01:
Hello @MariMakes,

Thanks for keeping this on the radar! Appreciate it!

I can confirm that the issue has been resolved after installing the build (#15098) that you suggested. This update helps our team be more productive in evaluating the STL files before committing to printing.

Regards.

P.S. Of course, I upgrade my desktop to Ubuntu 23.04 (and that may have newer drivers too!).

closed 2023-05-01 by baqwas