Ultimaker/Cura #14496: Slice of necklace with raft failed

jasbowen, 2023-02-07:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Anycubic Vyper

### Reproduction steps

(1) opened the file
(2) checked the settings, did not change anything because I tried slicing this before (the raft doesn't generate under the whole model, which I've already reported in issue# 00804185)
(3) Clicked on "Slice"
(4) it started to process, then an error dialog box popped up saying that the slicing encountered an unexpected error and that I should consider reporting it.

### Actual results

So far tonight I've been getting a sliced model with the raft only generating under a part of the model. After submitting the issue earlier this evening, I realized that the part of the model with the raft in the latest model file had a negative z-height, so I went back into TinkerCAD and adjusted the height of that object, then regrouped the model and set the z-height of the whole model to -0.01 to see if I could get the raft to propagate. Now I'm getting the slicing error.

### Expected results

I was hoping that the raft would generate for the entire model, however, perhaps I *am* getting the expected result for the z-height being -0.01. However, as this is related to part of an ongoing issue trying to slice this model with a raft, I thought I'd report it.
[cura.log](https://github.com/Ultimaker/Cura/files/10671164/cura.log)
[A4_Valentine Necklace with Clasp (1).zip](https://github.com/Ultimaker/Cura/files/10671166/A4_Valentine.Necklace.with.Clasp.1.zip)
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

again, I'm just reporting a slicer failure that occurred after trying to fix the larger issue for this model (the raft not generating under the whole model) in issue #00804185

GregValiant, 2023-02-09:
Thanks for the report and the file.
Here is the part in Mesh Mixer.  The magenta balls and surfaces are errors.  This model can't be expected to slice.  I don't know how you would print it either.  It's all small islands for most of the print.  It's a stringing nightmare.
[image]

Here is a report on the model.
-> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> **2 Non-manifold edges (?)
--> 1 Inverted faces (?)**
--> **302 Degenerate faces (?)**
**--> 4 Duplicate faces (?)**
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 36905ms ------
-> Vertex count changed from 96865 to 90538 (-6327)
-> Triangle count changed from 193532 to 183206 (-10326)

Here is the repaired model sliced in Cura and with a full raft.
[image]

I don't see a bug here.  That model is pretty bad but when repaired everything looks good.  I'm inclined to remove the Bug label from this and to close it.  I'll leave at is for a bit and you can let me know if you think that is premature.

jasbowen, 2023-02-09:
I'm pretty new to printing, so I didn't realize I imparted all of those
issues to the design. I appreciate your help very much. You can go ahead
and close it. Thank you!

Julie

GregValiant, 2023-02-09:
Cura can be complicated once you get into the advanced side.  I think CAD has a steeper learning curve though.
MS 3D Builder is pretty good at catching errors and repairing models.  It isn't a bad habit to run your models through it before you spend time trying to get a bad one to slice.
If you go to the Cura MarketPlace you can load the "Mesh Tools" plugin.  It has a bit of repair ability but it has a lot of nice functions for checking meshes and other things.
I'll remove the bug label and go ahead and close this.  Good luck with it.

closed 2023-02-09 by GregValiant