Ultimaker/Cura #14497: Cannot update "Super Quality" profile

JoshMcCullough, 2023-02-07:
### Application Version

5.2.1

### Platform

Manjaro

### Printer

N/A

### Reproduction steps

1. Load "Super Profile".
2. Make some change to the Print Settings.
3. Open the Profiles dialog.
4. Note that your change is listed, but the "Update profile" button is disabled.

### Bonus
The "Upddate profile." button text should not have a period at the end. And, IMO, it should be Title Case ("Update Profile") along with all other buttons, but that's an aside.

### Actual results

Cannot click to update the profile.

### Expected results

Can apply edits to the profile.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

JoshMcCullough, 2023-02-07:
While I was able to update the other profiles listed, it does not actually seem to "take". E.g. my `Z Offset` change does not persist as I move between profiles.

5axes, 2023-03-09:
Not a bug  Default profile cannot be updated as they are the reference for the other profile. You need to create a new one and manage your settings inside this new custom profile. Default profile can be change but you need for that to édit the configuration files . It's a job for expert users and unfortunatly this modification should be done on every new release.

MariMakes, 2023-05-05:
Hey @JoshMcCullough,

Sorry, it took us a while to get back to you 😞 
It's indeed a little confusing, but you can't update the default profiles. 
You can make a new one and update that one as much as you like. 

In Cura 5.3 we even made it easier by adding a save button where you can save a new one or overwrite an existing profile. 
[image]

To me, it doesn't seem like a bug in Ultimaker Cura, since we you are seeing expected behavior. 
Apologies for any inconvenience caused by our delayed response. 
I'll be closing this issue. I hope you understand.

closed 2023-05-05 by MariMakes

JoshMcCullough, 2023-05-06:
Okay, thanks. Not a bug per se but an interesting "gotcha" that might be worth explaining on the UI.