Ultimaker/Cura #14502: "Select face to align to build plate" is very broken.

RaptorUK, 2023-02-07:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

Drag the part into Cura.
Select the part of not already selected
Click - Rotate ("R")
Click - Select face to align to build plate
Select the face

### Actual results

Selected face is  NOT   aligned to build plate.

### Expected results

Selected face should be aligned to build plate.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

How do I give you the project file if you don't support  .3mf  files ? zip ?

[UMS5_VacuumStencil.Base.Cover.zip](https://github.com/Ultimaker/Cura/files/10676362/UMS5_VacuumStencil.Base.Cover.zip)

PC is a Dell Precision 7560 

By the way,  Lay Flat doesn't work for me either . . .

5axes, 2023-02-08:
Works Fine for me ... 

[image]

You can also use the plugin  AutoOrientation

RaptorUK, 2023-02-13:
I can't explain it . . . but its working for me today also.

closed 2023-02-13 by RaptorUK