Ultimaker/Cura #14503: Auto scale not correct

seltzdesign, 2023-02-07:
### Application Version

5.2.2

### Platform

Windows 10

### Printer

FLSun V400

### Reproduction steps

1. Import a model
2. Watch it autoscale to some random value, usually way too small

[image]

### Actual results

Model is scaled to some arbitrary tiny size

### Expected results

Model is scaled to maximum size.

This used to work just fine with the same printer in Cura 5.0.0

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10678323/cura.log)

MariMakes, 2023-02-15:
Hey @seltzdesign,

Welcome to the Ultimaker Cura Github 🚀 
That looks tiny 🐜 

Do you know which units have been used to export this STL?
In my experience exporting in MMs never brings problems.

Do you have a project file for us? 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

seltzdesign, 2023-02-15:
Hey @MariMakes 

The .stl files have all been exported the same way from Rhino 7 using mm as units. They are 3d scans of humans and are at actual size in Rhino, so around 1600 - 1800 mm tall.

Importing the .stl files they are all scaled at arbitrary sizes, but usually in the range of 0.1 - 2% size, even though the V400's print height is 400mm, which means you can scale the figures to around 20% easily.

Auto-scaling to maximum size always used to work fine in older versions.

I'll attach the .stl that is atuoscaled to 0.12% and the project file.

[230206_Woman standing 07.zip](https://github.com/Ultimaker/Cura/files/10742330/230206_Woman.standing.07.zip)

MariMakes, 2023-02-21:
Hey @seltzdesign,

I brought your issue up with the team. 
The Auto Scaling feature is in place to help people who exported their files in the wrong format. 
It makes mm from meters and kilometers, for example for architects. 
Cura seems to be doing exactly what it is told. 

[image]

How do you expect this feature to work?
Can we make the description more clear to reflect this correctly?

seltzdesign, 2023-02-21:
Hey @MariMakes 

Auto Scaling to me suggests it will automatically scale down larger models to exactly fit the print volume. That is how it has worked in the past.

Also my models ARE in mm, so are you suggesting that is wrong? The models are however in life size, so the human figures are around 1700mm large. So actually the unit has nothing to do with it, but the actual size of the model.

I just gave it another go now and it seems to be working fine, but it does fail quite often and it scales the models down to something really tiny as in the screenshot, usually well below 5% (even though normally the models fit at around 20%). If I can reproduce the exact circumstances where it happens, I will let you know.

modismo, 2023-05-04:
So yeah, this is still exactly the same behavior in 5.3.1. You would think this is a pretty easy fix. 
Step 1 - check size of geometry
Step 2 - if size > build plate size -> scale, else > don't do any scaling.