Ultimaker/Cura #14504: Cannot run Cura from Source: No module named 'PyQt6'

catapulta, 2023-02-07:
### Application Version

5.2.2

### Platform

macOS Monterrey (M1)

### Printer

NA

### Reproduction steps

On a clean environment:
1. Follow instructions in https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source#macos

2. Fails when running:
```
conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv
```

### Actual results

```
----Running------
> DYLD_LIBRARY_PATH="" DYLD_FRAMEWORK_PATH="" /Users/myusername/apps/Cura/venv/bin/python -m pip install -r /Users/myusername/apps/Cura/requirements.txt --upgrade
-----------------
Ignoring numpy: markers 'sys_platform == "win32"' don't match your environment
Ignoring twisted-iocpsupport: markers 'sys_platform == "win32"' don't match your environment
Ignoring pywin32: markers 'sys_platform == "win32"' don't match your environment
Ignoring pywin32-ctypes: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: PyQt6-sip==13.2.1 in ./venv/lib/python3.10/site-packages (from -r /Users/myusername/apps/Cura/requirements.txt (line 2)) (13.2.1)
Requirement already satisfied: PyQt6==6.2.3 in ./venv/lib/python3.10/site-packages (from -r /Users/myusername/apps/Cura/requirements.txt (line 7)) (6.2.3)
Requirement already satisfied: PyQt6-Qt6==6.2.4 in ./venv/lib/python3.10/site-packages (from -r /Users/myusername/apps/Cura/requirements.txt (line 12)) (6.2.4)
Collecting PyQt6-NetworkAuth==6.2.0
  Using cached PyQt6_NetworkAuth-6.2.0-cp36-abi3-macosx_10_14_universal2.whl (127 kB)
Collecting PyQt6-NetworkAuth-Qt6==6.2.4
  Using cached PyQt6_NetworkAuth_Qt6-6.2.4-py3-none-macosx_11_0_arm64.whl (77 kB)
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    PyQt6-NetworkAuth-Qt6==6.2.4 from https://files.pythonhosted.org/packages/17/0a/4dded3dde322915d532400de2284f5ff20cfa342d60aff8916b6fbf7d852/PyQt6_NetworkAuth_Qt6-6.2.4-py3-none-macosx_11_0_arm64.whl#sha256=1363ea81e5c6ac10bfd643e41ba0d215c0d031a57ff1e5972cc4c2a918efe712 (from -r /Users/myusername/apps/Cura/requirements.txt (line 22)):
        Expected sha256 1ae9e08e03bd9d5ebdb42dfaccf484a9cc62eeea7504621fe42c005ff1745e66
        Expected     or 8ed4e5e0eaaa42a6f91aba6745eea23fb3ffcbddc6b162016936530ed28dd0ad
        Expected     or c7996a9d8c4ce024529ec37981fbfd525ab1a2d497af1281f81f2b6054452d2e
             Got        1363ea81e5c6ac10bfd643e41ba0d215c0d031a57ff1e5972cc4c2a918efe712

WARNING: You are using pip version 22.0.4; however, version 23.0 is available.
You should consider upgrading via the '/Users/myusername/apps/Cura/venv/bin/python -m pip install --upgrade pip' command.
conanfile.py (cura/5.4.0-alpha): ERROR: Generator VirtualPythonEnv(file:None) failed
Error 1 while executing DYLD_LIBRARY_PATH="" DYLD_FRAMEWORK_PATH="" /Users/myusername/apps/Cura/venv/bin/python -m pip install -r /Users/myusername/apps/Cura/requirements.txt --upgrade
ERROR: Error 1 while executing DYLD_LIBRARY_PATH="" DYLD_FRAMEWORK_PATH="" /Users/myusername/apps/Cura/venv/bin/python -m pip install -r /Users/myusername/apps/Cura/requirements.txt --upgrade
```

### Expected results

No errors during compilation.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Happy to include full logs if useful-- I'd just need to re-run again.

catapulta, 2023-02-07:
After adding the right hash for `PyQt6-NetworkAuth` in `requirements.txt` (see diff below) and activating the venv, the error is:

```
Cura % python cura_app.py 
Traceback (most recent call last):
  File "/Users/aldo.marini/apps/Cura/cura_app.py", line 23, in <module>
    from PyQt6.QtNetwork import QSslConfiguration, QSslSocket
ModuleNotFoundError: No module named 'PyQt6'
```

I installed `PyQt6` via `pip` (successful) and then the error is:

```
Cura % python cura_app.py
Traceback (most recent call last):
  File "/Users/aldo.marini/apps/Cura/cura_app.py", line 25, in <module>
    from UM.Platform import Platform
ModuleNotFoundError: No module named 'UM'
```

`requirements.txt` diff:
```
diff --git a/requirements.txt b/requirements.txt
index 70a91f9a0e..4d73ae5fa6 100644
--- a/requirements.txt
+++ b/requirements.txt
@@ -22,6 +22,7 @@ PyQt6-NetworkAuth==6.2.0 \
 PyQt6-NetworkAuth-Qt6==6.2.4 \
     --hash=sha256:c7996a9d8c4ce024529ec37981fbfd525ab1a2d497af1281f81f2b6054452d2e \
     --hash=sha256:1ae9e08e03bd9d5ebdb42dfaccf484a9cc62eeea7504621fe42c005ff1745e66 \
+    --hash=sha256:1363ea81e5c6ac10bfd643e41ba0d215c0d031a57ff1e5972cc4c2a918efe712 \
     --hash=sha256:8ed4e5e0eaaa42a6f91aba6745eea23fb3ffcbddc6b162016936530ed28dd0ad
```

MariMakes, 2023-02-10:
Hey @catapulta,

😖 Sorry to hear you are stuck.

I'm only a fledgling member of the Build From Source Club 👑 
Before we get started, I have a few questions to narrow down the troubleshooting:

1. Can you make sure you have specifically Conan version 1.56.0?  
`pip install conan == 1.56.0 --force `
You can check the [Conan Documentation]( https://docs.conan.io/en/latest/installation.html ) for the latest version and installation instructions

2. Did you try clearing the Conan Cache? 
`conan remove "*" -f `

3. Can you make sure you have the latest config with this command?
`conan config install https://github.com/Ultimaker/conan-config.git`

4. Can you make sure you have a default profile?
`conan profile new default --detect --force`

5. With these steps you should have a clean slate and you can follow the steps on our wiki again. 
[You can find it here. ](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source) 

If you are stuck in any way, please reach out. We are happy to help out. 
The best way to do that is to provide us with your complete run-out log. 
Not just the part that fails, but all the lines generated after you've given your command. 
We need it to see your profiles and packages. 

Can you help me understand? What do you want to achieve by creating a Build?
Are you intending to share it with others or are you using it for research purposes or is it just for the fix for PyQT6?
I would love to know more about your project, maybe we can help.

catapulta, 2023-02-13:
Hi @MariMakes, thanks for checking in. The initial problem I had was related to PyQT6, but then found the root issue was related to the pip install (see title & description edit above).

**The fix is simple, please check out [PR](https://github.com/Ultimaker/Cura/pull/14506).**

To your questions:
1. Installed version is 1.56.02. 
```
Cura % conan --version
Conan version 1.56.0
```

2. Yes, I cleared the cache.
3. I use `conan config install https://github.com/Ultimaker/conan-config.git` every time after clearing the cache.
4. I also use `conan profile new default --detect --force` as a part of resetting `conan`.

jellespijker, 2023-02-14:
@MariMakes it seems that @catapulta fixed his own issue and made a contribution with the PR such that other people on MacOS Monterrey can now also setup Cura dev env on their computer.

closing this issue since I just merged your PR @catapulta let me know if you still run into problems

closed 2023-02-14 by jellespijker